ARMmbed/mbed-cli #401: How to compile application.hex for DFU / FOTA (nrf5x)?

jensstruemper, 2016-11-29:
Hi there, 
I was wondering if it is currently possible to compile an application only hex (without softdevice) for nrf5x targets using mbed-cli? If remember correctly there have been separate targets for FOTA in the past - but those don't exist anymore. 

Thanks,
Jens

bridadan, 2016-11-30:
@jensstruemper Although it's not really advertised, you should still be able to use the `*_OTA` targets which I believe don't contain the softdevice. So for instance if you're using the `NRF51_DK` target, you can use the `NRF51_DK_OTA` target instead for FOTA.

Full warning: I haven't tried this myself, but historically this has worked in the past I believe.

jensstruemper, 2016-11-30:
Thanks @bridadan! That works indeed. Just couldn't locate the *_OTA directory so I thought it wouldn't work anymore.

closed 2016-11-30 by jensstruemper