ARMmbed/mbed-cli #402: Building in debug mode does not filter the folders TARGET_{RELEASE,DEBUG}

AlessandroA, 2016-12-06:
Building an application with `--profile mbed-os/tools/profiles/debug.json` does not filter correctly for the `TARGET_RELEASE` and `TARGET_DEBUG` folders.

In particular:

* In a release build, `TARGET_RELEASE` is correctly included in the list of source folders, and `TARGET_DEBUG` is correctly filtered out.
* In a debug build, `TARGET_RELEASE` is incorrectly included in the list of source folders, `TARGET_DEBUG` is incorrectly filtered out.

It seems like the `RELEASE` label is applied regardless of the debug/release options.

Observed with mbed-cli 0.9.10 and 1.0.0.

@0xc0170 @meriac

0xc0170, 2016-12-06:
I am looking at this. Will report back my findings soon.

cc @theotherjimmy

0xc0170, 2016-12-06:
I can't reproduce the problem

Release profile:
```
[Error] test.cpp@1,0:  #35: #error directive: RELEASE
[ERROR] ".\TARGET_RELEASE\test.cpp", line 1: Error:  #35: #error directive: RELEASE
.\TARGET_RELEASE\test.cpp: 0 warnings, 1 error

[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u C:\2\mbed-os-example-blinky\mbed-os\tools\make.py -t ARM -m KL25Z --profile mbed-os/tools/profiles/default.json --source . --build .\BUILD\KL25Z\ARM" in "C:\2\mbed-os-example-blinky"
```

Debug profile:
```
[Error] test.cpp@1,0:  #35: #error directive: DEBUG
[ERROR] ".\TARGET_DEBUG\test.cpp", line 1: Error:  #35: #error directive: DEBUG
.\TARGET_DEBUG\test.cpp: 0 warnings, 1 error

[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u C:\2\mbed-os-example-blinky\mbed-os\tools\make.py -t ARM -m KL25Z --profile mbed-os/tools/profiles/debug.json --source . --build .\BUILD\KL25Z\ARM" in "C:\2\mbed-os-example-blinky"
```

0xc0170, 2016-12-06:
This is the bug in the mbed-os code base, here's a fix : https://github.com/ARMmbed/mbed-os/pull/3373

0xc0170, 2016-12-12:
This shall be closed?

AlessandroA, 2016-12-12:
Yes, fixed by https://github.com/ARMmbed/mbed-os/pull/3373. Thanks!

closed 2016-12-12 by AlessandroA