ARMmbed/mbed-cli #403: ERROR: Unable to write ignore file

evandavey, 2016-12-08:
OSX: 10.12.1
Python 2.7.12
Git: 2.11.0
mbed-cli: 1.0.0

```
mbed import mbed-os-example-blinky
[mbed] Importing program "mbed-os-example-blinky" from "https://github.com/ARMmbed/mbed-os-example-blinky" at latest revision in the current branch
[mbed] ERROR: Unable to write ignore file in "/Users/evandavey/Documents/engineering/mbed-os-example-blinky/.git/info/exclude"
```

bridadan, 2016-12-08:
@evandavey thanks for reporting this. Does this happen each time you run `mbed import mbed-os-example-blinky`? If so, could you please post the full verbose log by running `mbed import mbed-os-example-blinky -v`

evandavey, 2016-12-08:
Yes, every time.  Verbose log doesn't help much:

```
mbed] Working path "/Users/evandavey/Documents/engineering/experiments" (directory)
[mbed] Importing program "mbed-os-example-blinky" from "https://github.com/ARMmbed/mbed-os-example-blinky" at latest revision in the current branch
[mbed] Exec "git clone https://github.com/ARMmbed/mbed-os-example-blinky /Users/evandavey/Documents/engineering/experiments/mbed-os-example-blinky" in /Users/evandavey/Documents/engineering/experiments
Cloning into '/Users/evandavey/Documents/engineering/experiments/mbed-os-example-blinky'...
remote: Counting objects: 153, done.
remote: Total 153 (delta 0), reused 0 (delta 0), pack-reused 153
Receiving objects: 100% (153/153), 181.10 KiB | 171.00 KiB/s, done.
Resolving deltas: 100% (55/55), done.
[mbed] ERROR: Unable to write ignore file in "/Users/evandavey/Documents/engineering/experiments/mbed-os-example-blinky/.git/info/exclude"
```

bridadan, 2016-12-08:
Hmm, looks like it cloned correctly. Can you confirm that the directory `/Users/evandavey/Documents/engineering/experiments/mbed-os-example-blinky/.git/info/exclude` exists?

If so, what are the permissions on that folder? Can you run `ls -la` from `/Users/evandavey/Documents/engineering/experiments/mbed-os-example-blinky/.git/info`?

evandavey, 2016-12-08:
The info folder doesn't exist.  This is from `/Users/evandavey/Documents/engineering/experiments/mbed-os-example-blinky/.git/`

```
drwxr-xr-x  11 evandavey  staff   374  9 Dec 09:08 .
drwxr-xr-x   8 evandavey  staff   272  9 Dec 08:59 ..
-rw-r--r--   1 evandavey  staff    23  9 Dec 08:59 HEAD
-rw-r--r--   1 evandavey  staff   319  9 Dec 08:59 config
drwxr-xr-x   7 evandavey  staff   238  9 Dec 08:59 hooks
-rw-r--r--   1 evandavey  staff   477  9 Dec 09:07 index
drwxr-xr-x   4 evandavey  staff   136  9 Dec 08:59 logs
drwxr-xr-x   4 evandavey  staff   136  9 Dec 08:59 objects
-rw-r--r--   1 evandavey  staff  1725  9 Dec 08:59 packed-refs
drwxr-xr-x   5 evandavey  staff   170  9 Dec 08:59 refs
-rw-r--r--   1 evandavey  staff   370  9 Dec 08:59 tags
```

evandavey, 2016-12-08:
I tried the Apple version of git `git version 2.9.3 (Apple Git-75)` but same issue so that doesn't appear to be the problem.

evandavey, 2016-12-08:
1. Confirming `git init` does not create info folder.
2. Confirming `mbed deploy` reports same error
3. Confirming creating `.git/info` then `mbed deploy` works (but note the same write error occurs on the `mbed-os/.git/info` folder).

bridadan, 2016-12-08:
Great, thanks @evandavey! I've created a patch here: https://github.com/ARMmbed/mbed-cli/pull/404

Would you mind testing it locally to see if that fixes your problem?

bridadan, 2016-12-09:
Should be good to close when #404 is merged.

PR #404 (2016-12-09): Add 'info' folder to .git folder if missing (fixes #403)
Before writing the exclude file for the imported git repos, mbed-cli now checks if the info folder is there. If it isn't it now creates it before writing the file.

This should fix #403.
commit 90375fa3 (2016-12-09): Merge pull request #404 from bridadan/add_missing_git_folder

Add 'info' folder to .git folder if missing (fixes #403)
closed 2016-12-09 by screamerbg