ARMmbed/mbed-cli #406: compiler FAIL with NUCLEO_F302R8

fwndz, 2016-12-14:
Hi,

I got some problem when compiling my code by using NUCLEO_F302R8. The error shown below.
Building project SX1276PingPong (NUCLEO_F302R8, GCC_ARM)
```
Scan: .
Scan: mbed
Scan: env
Compile: radio.cpp
[Error] stm32f302x8.h@795,29: expected identifier before '(' token
[Error] stm32f302x8.h@795,43: expected ')' before '*' token
[Error] stm32f302x8.h@795,43: expected ')' before '*' token
[Error] radio.cpp@20,1: expected '}' at end of input
[ERROR] In file included from ./mbed/165afa46840b/TARGET_NUCLEO_F302R8/stm32f3xx.h:132:0,
                 from ./mbed/165afa46840b/TARGET_NUCLEO_F302R8/cmsis.h:35,
                 from ./mbed/165afa46840b/TARGET_NUCLEO_F302R8/TARGET_STM/TARGET_STM32F3/TARGET_NUCLEO_F302R8/objects.h:33,
                 from ./mbed/165afa46840b/TARGET_NUCLEO_F302R8/TARGET_STM/TARGET_STM32F3/TARGET_NUCLEO_F302R8/device.h:68,
                 from ./mbed/165afa46840b/platform.h:21,
                 from ./mbed/165afa46840b/mbed.h:21,
                 from ./SX1276Lib/radio/radio.h:18,
                 from ./SX1276Lib/radio/radio.cpp:15:
./mbed/165afa46840b/TARGET_NUCLEO_F302R8/stm32f302x8.h:795:29: error: expected identifier before '(' token
 #define CAN                 ((CAN_TypeDef *) CAN_BASE)
                             ^
./mbed/165afa46840b/CAN.h:67:7: note: in expansion of macro 'CAN'
 class CAN {
       ^
./mbed/165afa46840b/TARGET_NUCLEO_F302R8/stm32f302x8.h:795:43: error: expected ')' before '*' token
 #define CAN                 ((CAN_TypeDef *) CAN_BASE)
                                           ^
./mbed/165afa46840b/CAN.h:67:7: note: in expansion of macro 'CAN'
 class CAN {
       ^
./mbed/165afa46840b/TARGET_NUCLEO_F302R8/stm32f302x8.h:795:43: error: expected ')' before '*' token
 #define CAN                 ((CAN_TypeDef *) CAN_BASE)
                                           ^
./mbed/165afa46840b/CAN.h:67:7: note: in expansion of macro 'CAN'
 class CAN {
       ^
./SX1276Lib/radio/radio.cpp:20:1: error: expected '}' at end of input
 }
 ^

[mbed] ERROR: "python" returned error code 1.

```

Could you please help. When I do the the compilation in mbed os is fine.

bridadan, 2016-12-14:
Hi @fwndz, could you provide more details about your issue? Can you provide a link to your project? Can you also give us the version of mbed CLI (`mbed --version`) and your version of GCC (`arm-none-eabi-gcc --version`)?

fwndz, 2016-12-14:
hi @bridadan this is the version

```
 mbed --version
0.9.7

arm-none-eabi-gcc (15:4.9.3+svn231177-1) 4.9.3 20150529 (prerelease)
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
I applied this example https://developer.mbed.org/teams/Semtech/code/SX1276PingPong/ with NUCLEO_F302R8 that error occur.

fwndz, 2016-12-15:
Hi @bridadan  I already solved the issues . Thanks

closed 2016-12-15 by fwndz

bridadan, 2016-12-15:
@fwndz Sorry for my delayed response! Would you mind sharing the solution?

fwndz, 2016-12-17:
Hi @bridadan , I just replaced the given mbed library with the library from mbed cli.

bridadan, 2016-12-19:
Glad it worked out!