ARMmbed/mbed-cli #407: mbed add does not ignore "*.lib" files inside ".mbedignore" paths

bremoran, 2016-12-19:
I have a repo with both a KiCAD PCB design (which contains `*.lib` files) and a library for using the PCB in it. This causes a lot of spurious output from mbed add:

```
$ mbed add git@github.com:ARMmbed/power-cut-jig.git --protocol ssh
[mbed] Adding library "power-cut-jig.git" from "ssh://git@github.com/ARMmbed/power-cut-jig.git" at latest revision in the current branch
[mbed] WARNING: File "jig-b-cache.lib" in "/Users/bremor01/dev/experiments/addlib/power-cut-jig.git/jig-pcb" uses a non-standard .lib file extension, which is not compatible with the mbed build tools.
---
[mbed] WARNING: File "power-jig.lib" in "/Users/bremor01/dev/experiments/addlib/power-cut-jig.git/jig-pcb" uses a non-standard .lib file extension, which is not compatible with the mbed build tools.
---
[mbed] WARNING: File "jig-b-cache.lib" in "/Users/bremor01/dev/experiments/addlib/power-cut-jig.git/jig-pcb" uses a non-standard .lib file extension, which is not compatible with the mbed build tools.
---
[mbed] WARNING: File "power-jig.lib" in "/Users/bremor01/dev/experiments/addlib/power-cut-jig.git/jig-pcb" uses a non-standard .lib file extension, which is not compatible with the mbed build tools.
---
[mbed] WARNING: File "jig-b-cache.lib" in "/Users/bremor01/dev/experiments/addlib/power-cut-jig.git/jig-pcb" uses a non-standard .lib file extension, which is not compatible with the mbed build tools.
---
[mbed] WARNING: File "power-jig.lib" in "/Users/bremor01/dev/experiments/addlib/power-cut-jig.git/jig-pcb" uses a non-standard .lib file extension, which is not compatible with the mbed build tools.
---
[mbed] Updating reference "power-cut-jig.git" -> "https://github.com/ARMmbed/power-cut-jig/#416de319667fbfacfc12f858e85104d2f25bb78f"
```

bridadan, 2016-12-19:
Hmmm, maybe we should make these warning messages appear only when verbose (`-v`) is enabled. Any thoughts @screamerbg, @bremoran ?

bremoran, 2016-12-19:
@bridadan it would be less of a problem if the `.mbedignore` were respected by all of the `mbed *` commands.

seppestas, 2017-08-08:
Having the mbed library commands respect what's in the .mbedignore file would also solve some issues with these commands pulling in mbed-os in libraries, see #472 and #473.

In [my example](https://github.com/ARMmbed/mbed-cli/issues/472#issuecomment-298956337), having `mbed add` and `mbed deploy` respect the contents of .mbedignore would make it possible to add e.g an EXAMPLE folder to a library and put "EXAMPLE/*" in the .mbedignore file preventing libraries in the EXAMPLE folder (like mbed-os) from being pulled in and compiled when using the library.

In this case, if someone wants to run the example for testing / developing, it would be possible to simply `cd` into the EXAMPLE folder and run the mbed commands there, causing it not to see the .mbedignore file at the root of the library.

closed 2019-10-31 by ciarmcom