ARMmbed/mbed-cli #408: -fpermissive issues

fwndz, 2016-12-29:
Hi,

I have some issues when compiling the example code. When compiling with online compiler, the compilation is fine but when I try to use mbed cli there are some issues as shown below.
```
`Building project ESP8266_MQTT_HelloWorld (NUCLEO_F302R8, GCC_ARM)
Scan: .
Scan: FEATURE_LWIP
Scan: FEATURE_BLE
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_UVISOR
Scan: FEATURE_STORAGE
Scan: FEATURE_NANOSTACK
Scan: FEATURE_THREAD_ROUTER
Scan: FEATURE_LOWPAN_ROUTER
Scan: FEATURE_THREAD_END_DEVICE
Scan: FEATURE_NANOSTACK_FULL
Scan: FEATURE_LOWPAN_BORDER_ROUTER
Scan: FEATURE_LOWPAN_HOST
Scan: FEATURE_THREAD_BORDER_ROUTER
Scan: mbed
Scan: env
Compile [  0.4%]: ESP8266Interface.cpp
[Error] Endpoint.h@69,15: declaration of 'ESP8266* Endpoint::ESP8266' [-fpermissive]
[Error] ESP8266.h@41,7: changes meaning of 'ESP8266' from 'class ESP8266' [-fpermissive]
[ERROR] In file included from ./ESP8266Interface/ESP8266Interface.h:24:0,
                 from ./ESP8266Interface/ESP8266Interface.cpp:1:
./ESP8266Interface/Socket/Endpoint.h:69:15: error: declaration of 'ESP8266* Endpoint::ESP8266' [-fpermissive]
     ESP8266 * ESP8266;
               ^
In file included from ./ESP8266Interface/ESP8266Interface.h:23:0,
                 from ./ESP8266Interface/ESP8266Interface.cpp:1:
./ESP8266Interface/ESP8266/ESP8266.h:41:7: error: changes meaning of 'ESP8266' from 'class ESP8266' [-fpermissive]
 class ESP8266
       ^

[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u /home/fawwaz/mbedcli/ESP8266_MQTT_HelloWorld/mbed-os/tools/make.py -t GCC_ARM -m NUCLEO_F302R8 --source . --build ./BUILD/NUCLEO_F302R8/GCC_ARM" in "/home/fawwaz/mbedcli/ESP8266_MQTT_HelloWorld"
---
`
```

Can I know what kind of toolchain that is used in online mbed compiler? Why this problem occur?What -fpermissive mean?

Thanks.

Regards
fwndz

bridadan, 2016-12-29:
Hi @fwndz, could please provide the following information?

1. The repository you're trying to compile (if its available)
2. Post the verbose log when you compile (`mbed compile -v`)
3. Post the output of `pip list`
4. And finally, the version of GCC_ARM you're using (`arm-none-eabi-gcc --version`)

Thank you!

fwndz, 2016-12-29:
hi @bridadan 

1. I try to compile this repo https://developer.mbed.org/teams/ESP8266/code/ESP8266_MQTT_HelloWorld
2. The verbose log
```
Building project ESP8266_MQTT_HelloWorld (NUCLEO_F302R8, GCC_ARM)
Scan: .
Scan: FEATURE_LWIP
Scan: FEATURE_BLE
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_UVISOR
Scan: FEATURE_STORAGE
Scan: FEATURE_NANOSTACK
Scan: FEATURE_THREAD_ROUTER
Scan: FEATURE_LOWPAN_ROUTER
Scan: FEATURE_THREAD_END_DEVICE
Scan: FEATURE_NANOSTACK_FULL
Scan: FEATURE_LOWPAN_BORDER_ROUTER
Scan: FEATURE_LOWPAN_HOST
Scan: FEATURE_THREAD_BORDER_ROUTER
Scan: mbed
Scan: env
[DEBUG] Macros: -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DDEVICE_PORTOUT=1 -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -DTARGET_FF_MORPHO -DDEVICE_LOWPOWERTIMER=1 -DDEVICE_RTC=1 -DTOOLCHAIN_object -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DTOOLCHAIN_GCC -DDEVICE_STDIO_MESSAGES=1 -DDEVICE_CAN=1 -DTARGET_CORTEX_M -DDEVICE_I2C_ASYNCH=1 -DTARGET_LIKE_CORTEX_M4 -DDEVICE_ANALOGOUT=1 -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DDEVICE_SPI_ASYNCH=1 -DDEVICE_SERIAL=1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_I2C=1 -DTRANSACTION_QUEUE_SIZE_SPI=2 -D__CORTEX_M4 -DTARGET_NUCLEO_F302R8 -DTARGET_STM32F302R8 -D__FPU_PRESENT=1 -DTARGET_FF_ARDUINO -DDEVICE_PORTIN=1 -DTARGET_RELEASE -DTARGET_STM -DDEVICE_SERIAL_FC=1 -DMBED_BUILD_TIMESTAMP=1483028674.24 -D__MBED_CMSIS_RTOS_CM -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DDEVICE_SPI=1 -DDEVICE_SPISLAVE=1 -DTARGET_STM32F3 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DARM_MATH_CM4
Compile [  2.2%]: Endpoint.cpp
[DEBUG] Compile: /usr/bin/arm-none-eabi-g++ -std=gnu++98 -fno-rtti -Wvla -c -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -Os -DMBED_RTOS_SINGLE_THREAD -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DDEVICE_PORTOUT=1 -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -DTARGET_FF_MORPHO -DDEVICE_LOWPOWERTIMER=1 -DDEVICE_RTC=1 -DTOOLCHAIN_object -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DTOOLCHAIN_GCC -DDEVICE_STDIO_MESSAGES=1 -DDEVICE_CAN=1 -DTARGET_CORTEX_M -DDEVICE_I2C_ASYNCH=1 -DTARGET_LIKE_CORTEX_M4 -DDEVICE_ANALOGOUT=1 -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DDEVICE_SPI_ASYNCH=1 -DDEVICE_SERIAL=1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_I2C=1 -DTRANSACTION_QUEUE_SIZE_SPI=2 -D__CORTEX_M4 -DTARGET_NUCLEO_F302R8 -DTARGET_STM32F302R8 -D__FPU_PRESENT=1 -DTARGET_FF_ARDUINO -DDEVICE_PORTIN=1 -DTARGET_RELEASE -DTARGET_STM -DDEVICE_SERIAL_FC=1 -DMBED_BUILD_TIMESTAMP=1483028674.24 -D__MBED_CMSIS_RTOS_CM -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DDEVICE_SPI=1 -DDEVICE_SPISLAVE=1 -DTARGET_STM32F3 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DARM_MATH_CM4 @./BUILD/NUCLEO_F302R8/GCC_ARM/.includes_51453ef273c4a51a51fdb317b7ac2cea.txt -include ./BUILD/NUCLEO_F302R8/GCC_ARM/mbed_config.h -MD -MF ./BUILD/NUCLEO_F302R8/GCC_ARM/ESP8266Interface/Socket/Endpoint.d -o ./BUILD/NUCLEO_F302R8/GCC_ARM/ESP8266Interface/Socket/Endpoint.o ./ESP8266Interface/Socket/Endpoint.cpp
[Error] Socket.h@25,7: redefinition of 'class Socket'
[Error] Socket.h@32,7: previous definition of 'class Socket'
[Error] Endpoint.h@69,15: declaration of 'ESP8266* Endpoint::ESP8266' [-fpermissive]
[Error] ESP8266.h@41,7: changes meaning of 'ESP8266' from 'class ESP8266' [-fpermissive]
[DEBUG] Return: 1
[DEBUG] Output: In file included from ./ESP8266Interface/Socket/Endpoint.cpp:18:0:
[DEBUG] Output: ./ESP8266Interface/Socket/Socket.h:25:7: error: redefinition of 'class Socket'
[DEBUG] Output:  class Socket {
[DEBUG] Output:        ^
[DEBUG] Output: In file included from ./mbed-os/features/netsocket/nsapi.h:39:0,
[DEBUG] Output:                  from ./mbed-os/mbed.h:43,
[DEBUG] Output:                  from ./ESP8266Interface/ESP8266/ESP8266.h:30,
[DEBUG] Output:                  from ./ESP8266Interface/Socket/Socket.h:21,
[DEBUG] Output:                  from ./ESP8266Interface/Socket/Endpoint.cpp:18:
[DEBUG] Output: ./mbed-os/features/netsocket/Socket.h:32:7: error: previous definition of 'class Socket'
[DEBUG] Output:  class Socket {
[DEBUG] Output:        ^
[DEBUG] Output: In file included from ./ESP8266Interface/Socket/Endpoint.cpp:19:0:
[DEBUG] Output: ./ESP8266Interface/Socket/Endpoint.h:69:15: error: declaration of 'ESP8266* Endpoint::ESP8266' [-fpermissive]
[DEBUG] Output:      ESP8266 * ESP8266;
[DEBUG] Output:                ^
[DEBUG] Output: In file included from ./ESP8266Interface/Socket/Socket.h:21:0,
[DEBUG] Output:                  from ./ESP8266Interface/Socket/Endpoint.cpp:18:
[DEBUG] Output: ./ESP8266Interface/ESP8266/ESP8266.h:41:7: error: changes meaning of 'ESP8266' from 'class ESP8266' [-fpermissive]
[DEBUG] Output:  class ESP8266
[DEBUG] Output:        ^
Traceback (most recent call last):
  File "/home/fawwaz/mbedcli/ESP8266_MQTT_HelloWorld/mbed-os/tools/make.py", line 293, in <module>
    toolchain))
  File "/home/fawwaz/mbedcli/ESP8266_MQTT_HelloWorld/mbed-os/tools/build_api.py", line 452, in build_project
    resources.inc_dirs)
  File "/home/fawwaz/mbedcli/ESP8266_MQTT_HelloWorld/mbed-os/tools/toolchains/__init__.py", line 807, in compile_sources
    return self.compile_queue(queue, objects)
  File "/home/fawwaz/mbedcli/ESP8266_MQTT_HelloWorld/mbed-os/tools/toolchains/__init__.py", line 870, in compile_queue
    raise ToolException(err)
ToolException: In file included from ./ESP8266Interface/Socket/Endpoint.cpp:18:0:
[... cut ...]
Tempita (0.5.2)
tortoisehg (3.7.3)
unity-lens-photos (1.0)
virtualenv (15.0.3)
Werkzeug (0.11.11)
wheel (0.29.0)
WTForms (2.1)
wxPython (3.0.2.0)
wxPython-common (3.0.2.0)
xmltodict (0.10.2)
```
4.Arm gcc version
```
arm-none-eabi-gcc (15:4.9.3+svn231177-1) 4.9.3 20150529 (prerelease)
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
Thank you

bridadan, 2016-12-29:
Thanks for providing that info. Didn't you make a similar issue for this particular example? https://github.com/ARMmbed/mbed-cli/issues/358. Are you trying anything new here? I think we determined that this example had only been tested in the online compiler which uses the ARM Compiler 5 toolchain. I don't think this has been tested with GCC_ARM.

fwndz, 2016-12-29:
Hi @bridadan. This is `mbed compile -v`

```
[mbed] Working path "D:\mbed\ESP8266_MQTT_HelloWorld" (program)
[mbed] Exec "python -u D:\mbed\ESP8266_MQTT_HelloWorld\mbed-os\tools\make.py -t ARM -m NUCLEO_F302R8 --source . --build .\BUILD\NUCLEO_F302R8\ARM -v" in D:\mbed\ESP8266_MQTT_HelloWorld
Building project ESP8266_MQTT_HelloWorld (NUCLEO_F302R8, ARM)
Scan: .
Scan: FEATURE_BLE
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_LWIP
Scan: FEATURE_UVISOR
Scan: FEATURE_LOWPAN_BORDER_ROUTER
Scan: FEATURE_LOWPAN_HOST
Scan: FEATURE_LOWPAN_ROUTER
Scan: FEATURE_NANOSTACK
Scan: FEATURE_NANOSTACK_FULL
Scan: FEATURE_THREAD_BORDER_ROUTER
Scan: FEATURE_THREAD_END_DEVICE
Scan: FEATURE_THREAD_ROUTER
Scan: FEATURE_STORAGE
Warning: Multiple linker scripts detected: .\mbed\cbbeb26dbd92\TARGET_NUCLEO_F302R8\TOOLCHAIN_ARM_STD\stm32f302x8.sct -> .\mbed-os\targets\TARGET_STM\TARGET_STM32F3\TARGET_NUCLEO_F302R8\device\TOOLCHAIN_AR
t
Scan: mbed
Scan: env
[DEBUG] Macros: -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DDEVICE_PORTOUT=1 -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -DTARGET_FF_MORPHO -DDEVICE_LOWPOWERTIMER=1 -DDEVICE_RTC=1 -DTOOLCHAIN_object
NCH=1 -D__CMSIS_RTOS -DMBED_USERNAME=fwndz -DDEVICE_STDIO_MESSAGES=1 -DDEVICE_CAN=1 -DTARGET_CORTEX_M -DTARGET_LIKE_CORTEX_M4 -DDEVICE_ANALOGOUT=1 -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DDEVICE_SPI_ASYNC
1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_I2C=1 -DTRANSACTION_QUEUE_SIZE_SPI=2 -D__CORTEX_M4 -DTARGET_NUCLEO_F302R8 -DTARGET_STM32F302R8 -D__FPU_PRESENT=1 -DTARGET_FF_ARDUINO -DDEVICE_PORTIN=1 -DTARGET_RELEASE -DT
ERIAL_FC=1 -D__MBED_CMSIS_RTOS_CM -DDEVICE_SLEEP=1 -DDEVICE_SPI=1 -DTOOLCHAIN_ARM_STD -DDEVICE_SPISLAVE=1 -DTARGET_STM32F3 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DTOOLCHAIN_ARM -DARM_MATH_CM4 -DMBED_BUILD_
.45
Compile [  1.1%]: Endpoint.cpp
[DEBUG] Compile: C:/Keil_v5/ARM/ARMCC\bin\armcc -c --gnu -Otime --split_sections --apcs=interwork --brief_diagnostics --restrict --multibyte_chars -O3 --cpu=Cortex-M4.fp --md --no_depend_system_headers --c
cpp --no_rtti --no_vla -I "C:/Keil_v5/ARM/ARMCC\include" -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DDEVICE_PORTOUT=1 -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -DTARGET_FF_MORPHO -DDEVICE_LOWPOWER
=1 -DTOOLCHAIN_object -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DMBED_USERNAME=fwndz -DDEVICE_STDIO_MESSAGES=1 -DDEVICE_CAN=1 -DTARGET_CORTEX_M -DTARGET_LIKE_CORTEX_M4 -DDEVICE_ANALOGOUT=1 -DTARGET_M4 -DTAR
ED -DDEVICE_SPI_ASYNCH=1 -DDEVICE_SERIAL=1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_I2C=1 -DTRANSACTION_QUEUE_SIZE_SPI=2 -D__CORTEX_M4 -DTARGET_NUCLEO_F302R8 -DTARGET_STM32F302R8 -D__FPU_PRESENT=1 -DTARGET_FF_ARDUI
 -DTARGET_RELEASE -DTARGET_STM -DDEVICE_SERIAL_FC=1 -D__MBED_CMSIS_RTOS_CM -DDEVICE_SLEEP=1 -DDEVICE_SPI=1 -DTOOLCHAIN_ARM_STD -DDEVICE_SPISLAVE=1 -DTARGET_STM32F3 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DT
ATH_CM4 -DMBED_BUILD_TIMESTAMP=1483004378.45 --via .\BUILD\NUCLEO_F302R8\ARM\.includes_89bdc5d6a11e04537d2a185fc24b5de3.txt --preinclude=.\BUILD\NUCLEO_F302R8\ARM\mbed_config.h --depend .\BUILD\NUCLEO_F302
ace\Socket\Endpoint.d -o .\BUILD\NUCLEO_F302R8\ARM\ESP8266Interface\Socket\Endpoint.o .\ESP8266Interface\Socket\Endpoint.cpp
[Warning] ESP8266.h@221,0:  #1-D: last line of file ends without a newline
[Error] Socket.h@25,0:  #256: invalid redeclaration of type name "Socket" (declared at line 32 of "./mbed-os/features/netsocket/Socket.h")
[DEBUG] Return: 1
[DEBUG] Output: "./ESP8266Interface/ESP8266/ESP8266.h", line 221: Warning:  #1-D: last line of file ends without a newline
[DEBUG] Output: "./ESP8266Interface/Socket/Socket.h", line 25: Error:  #256: invalid redeclaration of type name "Socket" (declared at line 32 of "./mbed-os/features/netsocket/Socket.h")
[DEBUG] Output: .\ESP8266Interface\Socket\Endpoint.cpp: 1 warning, 1 error
Traceback (most recent call last):
  File "D:\mbed\ESP8266_MQTT_HelloWorld\mbed-os\tools\make.py", line 293, in <module>
    toolchain))
  File "D:\mbed\ESP8266_MQTT_HelloWorld\mbed-os\tools\build_api.py", line 452, in build_project
    resources.inc_dirs)
  File "D:\mbed\ESP8266_MQTT_HelloWorld\mbed-os\tools\toolchains\__init__.py", line 807, in compile_sources
    return self.compile_queue(queue, objects)
  File "D:\mbed\ESP8266_MQTT_HelloWorld\mbed-os\tools\toolchains\__init__.py", line 870, in compile_queue
    raise ToolException(err)
ToolException: "./ESP8266Interface/ESP8266/ESP8266.h", line 221: Warning:  #1-D: last line of file ends without a newline
"./ESP8266Interface/Socket/Socket.h", line 25: Error:  #256: invalid redeclaration of type name "Socket" (declared at line 32 of "./mbed-os/features/netsocket/Socket.h")
.\ESP8266Interface\Socket\Endpoint.cpp: 1 warning, 1 error

[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u D:\mbed\ESP8266_MQTT_HelloWorld\mbed-os\tools\make.py -t ARM -m NUCLEO_F302R8 --source . --build .\BUILD\NUCLEO_F302R8\ARM -v" in "D:\mbed\ESP8266_MQTT_HelloWorld"
---
```

0xc0170, 2016-12-30:
Just to be certain - the program you referenced is including mbed.bld, is that mbed 2 library? I would say so. There is only one Socket class definition, that's in ESP library.

Using mbed-cli, there's another Socket class definition in networksocket folder. That would be my assumption based on the outputs shared above.

bridadan, 2017-01-03:
Ah yes, @0xc0170 is right. From the latest debug log you posted, it looks like you have both an mbed 2 build and mbed OS in your project. This line in particular is telling:

```
Warning: Multiple linker scripts detected: .\mbed\cbbeb26dbd92\TARGET_NUCLEO_F302R8\TOOLCHAIN_ARM_STD\stm32f302x8.sct -> .\mbed-os\targets\TARGET_STM\TARGET_STM32F3\TARGET_NUCLEO_F302R8\device\TOOLCHAIN_AR
```

fwndz, 2017-01-15:
@bridadan @0xc0170  thanks for your supports, I already solve the problem. There are some conflict between existing socket library with the library provided by mbed-os.

closed 2017-01-15 by fwndz