ARMmbed/mbed-cli #409: Add a 'mbed compile --debug' command

janjongboom, 2017-01-05:
It's hard to figure out how to do a debug build using mbed CLI, instructions differ between mbed OS 2.0, 5.0/5.1, 5.2+... mbed CLI should have a command to make such a build.

FYI, here are the instructions:

* If using mbed 2.0, include `-o debug-info` in your `workspace_tools/build.py` and `workspace_tools/make.py` commands to build with debug symbols.
* If using mbed 5.1, add `-o debug-info` to your `mbed compile` command.
* If using mbed 5.2+, add `--profile mbed-os/tools/profiles/debug.json` to your `mbed compile` command.

bridadan, 2017-01-05:
I think as of 5.3, we simplified it a bit so you can just do `--profile debug`. Do you think this is enough?

janjongboom, 2017-01-05:
@bridadan But that only works for 5.3... We should have a standard way of doing this across all versions of mbed OS.

0xc0170, 2017-01-06:
> @bridadan But that only works for 5.3... We should have a standard way of doing this across all versions of mbed OS.

What do you suggest? The way to go are profiles. This was a breaking change, and a replacement for old ``-o`` options that were not much clear what they do.

janjongboom, 2017-01-06:
@0xc0170 Sure, but I think making debug builds is such a common action that we should add it to mbed CLI. Suggestion:

```
mbed build --debug
```

Which magically sets some stuff in mbed 2, builds with the right `-o` in mbed OS 5.2 and lower, and with the debug profile in mbed OS 5.3.

bridadan, 2017-01-06:
I see what you're saying @janjongboom, though there's a part of me that doesn't want to treat debug as "special" even though it is such a common operation.

@screamerbg do you have any thoughts on this?

screamerbg, 2018-05-26:
I'll second @bridadan. @janjongboom I'll close the issue as there wasn't any activity for some time, but please reopen or create new with proposal how this could work as a workflow.
@BlackstoneEngineering

closed 2018-05-26 by screamerbg