ARMmbed/mbed-cli #416: mbed export -i eclipse_gcc_arm (with TARGET=K64F) results in a less functional eclipse project

jahess, 2017-01-07:
Presuming eclipse_gcc_arm refers to http://gnuarmeclipse.github.io, the generated project does not have target information set properly.  Specifically (and with a FRDM-K64F in mind):

mbed import https://github.com/ARMmbed/mbed-os-example-blinky blinky_test
cd blinky_test
mbed config TOOLCHAIN GCC_ARM
mbed config TARGET K64F
mbed export -i eclipse_gcc_arm

Results in a blinky_test->Properties->C/C++ Build with tabs of only:

  Devices|Binary Parsers|Error Parsers

rather than the more typical:

  Tool Setting|Toolchains|Devices|Binary Parsers|Error Parsers

ilg-ul, 2017-01-07:
the Tool Settings and Toolchains tabs are specific to managed GNU ARM Eclipse projects.

the `mbed export` should have an explicit option to generate managed GNU ARM Eclipse projects for your project to take advantage of the GNU ARM Eclipse features.

> Presuming eclipse_gcc_arm refers to http://gnuarmeclipse.github.io, 

unfortunately this is not true.

0xc0170, 2017-01-09:
cc @theotherjimmy @sarahmarshy

theotherjimmy, 2017-01-09:
@jahess @ilg-ul ATM the Eclipse exporters are generating a project file that simply calls out to a makefile. I think that's why you are missing the toolchain related tabs. Using the built in CDT builder would require a few things:
 * Documentation of the file format
 * a method to build the projects without going through the GUI for CI

If you have links for any of those things, I may be able to expedite the process of rewriting the CDT exporter to use the Eclipse native one.

ilg-ul, 2017-01-10:
@theotherjimmy,

generating GNU ARM Eclipse managed projects is just a matter of two files, .project and .cproject.

the format is not formally documented, but it is exactly the same as used by all managed Eclipse CDT projects, to which GNU ARM Eclipse adds some extensions.

mBed already exports CDT managed projects for other Eclipse based IDEs, like 'SW4STM32 System Workbench for STM32', and probably 'Kinetis Design Studio' and DF-5.

'Kinetis Design Studio' might be of special interest, since it uses an old version of the GNU ARM Eclipse plug-ins, so I would start from it and update it for the most recent version.

ilg-ul, 2017-01-10:
@theotherjimmy @sarahmarshy

is the code used to export for other IDEs publicly available? perhaps I can contribute to the GNU ARM Eclipse exporter.

theotherjimmy, 2017-01-10:
Yes! It's it the tools/export subdirectory of https://github.com/ARMmbed/mbed-os

theotherjimmy, 2017-01-10:
A quick crash course on the API there:
1) Exporter Class (to be extended) https://github.com/ARMmbed/mbed-os/blob/master/tools/export/exporters.py
2) The `self.toolchain` and `self.resources` attributes are the `mbedToolchain` and `Resources` objects found here https://github.com/ARMmbed/mbed-os/blob/master/tools/toolchains/__init__.py
3) We use Jinja 2 for the templates http://jinja.pocoo.org/docs/dev/templates/

0xc0170, 2017-01-10:
> is the code used to export for other IDEs publicly available? perhaps I can contribute to the GNU ARM Eclipse exporter.

Awesome!

Thanks @theotherjimmy for all the pointers here.

ilg-ul, 2017-01-10:
> Awesome!

:-)

I'm already creating the templates for GNU ARM Eclipse. The initial approach to reuse KDS files failed, the KDS templates do not generate UIDs properly.

 @theotherjimmy, is there any way I can test the exporters?

theotherjimmy, 2017-01-10:
@ilg-ul Yes.

The setup I use is:
 1) import a `mbed-os-example-*` project (https://github.com/ARMmbed/mbed-os-example-blinky)
 2) Replace the mbed-os folder with a symlink to my development copy of mbed-os
 3) `mbed export -i eclipse_gcc_arm -m <target>`
 4) Open the project in eclipse

Step 2 might be hard on windows.

theotherjimmy, 2017-01-10:
When you want any review of your template and associated python code, just create a Pull Request on Github and mention me somewhere. I'm happy to review!

ilg-ul, 2017-01-10:
https://github.com/ARMmbed/mbed-os/pull/3561

unfortunately the pull request requires more than a review, there are several issues that need to be fixed.

theotherjimmy, 2017-01-10:
That's fine. We can work out the problems with it.

screamerbg, 2017-03-23:
@theotherjimmy @ilg-ul Is this still an issue?

ilg-ul, 2017-03-23:
I don't know, I opened a separate bug for it https://github.com/ARMmbed/mbed-os/issues/3964

screamerbg, 2017-06-21:
@ilg-ul @theotherjimmy Is this still an issue?

ilg-ul, 2017-06-21:
when I checked, some time ago, it was ok, but I do not know with the current version. why? did you notice anything unusual?

screamerbg, 2017-06-21:
Just wanted to confirm the status and close it. Thanks.

closed 2017-06-21 by screamerbg