ARMmbed/mbed-cli #418: mbed add changes used protocol

jokimies, 2017-01-10:
We have some libraries in private repository, where there's no anonymous
users but everyone need to authenticate. For the authentication ssh keys has
been setup and basically all communication to the repositories happens via
``ssh``. 

Hence we add our libraries with

    mbed add git@git.ourdomain.com:ourlibrary

The library is added fine, but when writing the `.lib` file ``mbed``
alters the protocol from ``ssh`` to ``https`` and file ``ourlibrary.lib``
ends up having

    https://git.ourdomain.com/ourlibrary

Next time user writes ``mbed update`` it goes to
``https://git.ourdomain.com/ourlibrary`` and as authentication is required,
credentials are prompted. For user this rather inconvenient. But more
importanly, this stops automated build as the buildbot is prompted for
credentials. 

There's a possibility to set default ``protocol`` to ``ssh``. Doing that
solves our private repository problem, but introduces a new one: we use
libraries from public repositories, such as ``Github`` as well, and ``ssh``
protocol is applied to those as well. As ``Github`` is not our primary
repository, most of the users don't have a ``Github`` account. That being
the case, users cannot retreive
e.g. ``ssh://git@github.com/ARMmbed/mbed-os.git`` (but ``https://github.com/ARMmbed/mbed-os/`` works)

To overcome the problem, we need to edit the generated ``ourlibrary.lib``
(even though the documentation advices against it!) and edit the
``https://git.ourdomain.com/ourlibrary#<sha>`` written by ``mbed`` and
replace it with ``ssh://git@git.ourdomain.com/ourlibrary#<sha>``.  That way
everything works as we like.

From our point of view, ``mdeb add`` should retain the protocol that was used
when library was added, and not alter it to ``https`` or anything else

Is there some other way to achieve the same goal, i.e. having repositories
being fetched using both ``ssh`` and ``https``.

If not, could that feature be added?

bridadan, 2017-01-12:
@screamerbg following up on the discussion in #400, not being able to pin down `https` or `ssh` for certain libraries can cause issues due to IT policies of certain companies.

Is it possible to add extra data to the `.lib` files to indicate that `ssh` should be used for that particular library? Or would there be other downsides to not changing the `ssh` to `https` by default in the `.lib` files?

bridadan, 2017-01-12:
cc @theotherjimmy @sg-

theotherjimmy, 2017-01-12:
I can't think of a reason that keeping the initial add's protocol would be bad. @screamerbg might be able to explain it though.

bridadan, 2017-01-12:
@theotherjimmy I can imagine there can be an inconvenience when two developers use different authentication methods for Git repositories that supports both `https` and `ssh`. This could result in developers constantly going back and forth, altering the `.lib` file.

Maybe we could add a line below the address to the `.lib` file. Something like:

```
https://github.com/ARMmbed/mbed-os/#2885c1b41e63158cb6faf5f107cd821ae06ef26c
FORCE_PROTOCOL=ssh
```

This could be added with an `mbed add --force-protocol=ssh <url>`. I believe this is different from the `--protocol` option, which just affects how the library is cloned that first time.

The `FORCE_PROTOCOL` option in the `.lib` should only be used for libraries that are available only on hosts that just support a certain protocol OR a team/company requires its developers to use a specific protocol due to network policies/limitations.

However this may affect existing tools/the online compiler @theotherjimmy?

betzw, 2017-01-13:
Why not applying the KISS principle and at least for the default cases (i.e. non special options, settings etc.) just maintain the protocol as specified at the beginning of the URL within the `.lib` file?

bridadan, 2017-01-13:
@betzw I was concerned about the following case:

1. A user has ssh keys configured for GitHub or any other Git host
2. That user uses `--protocol ssh` for everything
3. The user adds library via ssh, so the `.lib` is saved like `ssh://git@github.com/myorg/awesome-library.git`
    1. This is a public repository, so ssh isn't actually needed. It would work just fine with `https`
4. This user publishes this application on GitHub.
5. Another user who does not have ssh keys configured now tries to import this application, but this fails because the library attempts to clone over `ssh`. Even though its a public library and it would work just fine over `https`.

I believe this is why we default to `https`, since it will always work for public repositories, which should hopefully be most of the code people are using. I don't think we should break this behavior so we can accommodate the less common (but still valid) use case we're discussing here. The method I proposed would be a non-invasive way of preserving the "default to https behavior" and some of the benefits @screamerbg mentioned [here](https://github.com/ARMmbed/mbed-cli/issues/400#issuecomment-270344914).

I'm of course open to other suggestions though! Or perhaps I'm making an assumption somewhere, so feel free to poke holes in my concerns :smile:

betzw, 2017-01-18:
Well, as a user I personally would love if I could add libraries which are just available internally with `--protocol ssh` hoping that `mbed-cli` continues to maintain my choice (which is currently not the case, e.g. `mbed sync`). 
Once I publish (e.g. using `mbed publish`) my program or library maybe this would be the right moment for me of being warned by `mbed-cli` about the not usage of the default protocol `https` asking me (and not acting without my consent) what I intend to do: e.g. substitute the URL and/or protocol, maintain the non default protocol, etc, ...
What do you think @bridadan?

bridadan, 2017-01-18:
Ok, you're slowly wearing me down :smile:

I could see if a user specifically supplies `--protocol ssh` to `mbed add`, we could save it with `ssh://`. I would imagine people would only ever do this for private repos or in some of the situations we discussed above.

However, if a user has set their global protocol to `ssh` and then does an `mbed add` WITHOUT the explicit `--protocol ssh`, then what should it do? The case I want to avoid is saving a perfectly accessible public repo over `http` (like mbed OS) as `ssh://` and then have other people complain that it doesn't work because someone's global config is different from there's.

I would say in this case, mbed CLI should IGNORE the global protocol setting when saving the `.lib`. The only thing I don't like about this is this would now be the ONLY global config option that would behave differently from their non-global counterpart. It could lead to nasty surprises for some people. I think this is the main reason why I was arguing for an explicit `--force-protocol` or something equivalent. So perhaps the `FORCE_PROTOCOL` in the `.lib` is not necessary. But I still have concerns over the CLI behavior with the `--protocol` arguments and the globals.

Sorry for the wall of text! Is my concern clear? Do you have any thoughts? And thanks for being patient with this, I know it probably seems minor! But being able to reliably reproduce code is one of the main goals of mbed, so potentially breaking this with an authentication option just makes me extra cautious! :smile:

betzw, 2017-01-20:
> However, if a user has set their global protocol to  ssh  and then does an  mbed add  WITHOUT the explicit  --protocol ssh , then what should it do? The case I want to avoid is saving a perfectly [...]

Why should a user set its global protocol to `ssh`? Only at a point where he thinks for some reason all repos should be accessed with this protocol. Ok, but when he publishes he would in my proposal be warned about this anomaly.
But maybe I didn't chatch actually your concern ...

bridadan, 2017-01-20:
@betzw When working with a mix of public and private repositories this a very common thing. Otherwise you'd have to provide https credentials any time you do an `mbed update`, which is quite tedious!

>  Ok, but when he publishes he would in my proposal be warned about this anomaly.

This is assuming the developer use `mbed publish` to do this. It is actually quite fine to push with `git` or `hg`. I believe mbed CLI handles this quite well. I do it quite often in my workflow. But they wouldn't get a warning about the library being saved as `ssh`. This then would cause the scenario I described in my previous comment:

> However, if a user has set their global protocol to ssh and then does an mbed add WITHOUT the explicit --protocol ssh, then what should it do? The case I want to avoid is saving a perfectly [...]

betzw, 2017-01-24:
Hi @bridadan,
well, maybe I didn't catch entirely your point, but probably this is not a surprise as I am actually not deeply enough plunged into the `mbed-cli` tool. I just use it mainly for compiling and from time to time to maintain an example program etc.
What I can say is therefore limited to this kind of experience, in which in my eyes the `mbed-cli` tool or rather some of its commands (e.g. `mbed sync`) lack the possibility to express exactly what I want it to do as they are performing operations I didn't expect (e.g. modify the protocol in the `.lib` files) and I couldn't find any another way to get the behavior I needed as going to sync/update my `.lib` files manually.
This is not a drama but is for sure inconvenient ...
Cheers,
Wolfgang

betzw, 2017-01-24:
> However, if a user has set their global protocol to ssh and then does an mbed add WITHOUT the explicit --protocol ssh, then what should it do? The case I want to avoid is saving a perfectly [...]

Anyway, let me try to answer to this again: I would say that whenever a command is emitted without the explicit `--protocol` option you should maintain the protocol which you find in the `.lib` file. In case there is no `.lib` file yet, use the default/global protocol (whatever it is set to). 
Furthermore, I would consider the case where the default/global protocol is set to `ssh` and the user adds a perfectly accessible public repo over `http` without an explicit `--protocol http` as a mistake of the user. Remember, you cannot avoid that users commit mistakes however you design your tool :smile:

screamerbg, 2017-06-21:
@betzw I think that @bridadan explained it in quite a detail. Large IT companies/corporations have very strict IT rules about the use of SSH and connecting to external services. We want to avoid complaints like "mbed CLI made an SSH connection attempt to an external host, and I got a flagged from our IT department for violating the company IT policy".

closed 2017-06-21 by screamerbg