ARMmbed/mbed-cli #420: Empty project exported to SW4STM32

abouillot, 2017-01-14:
When I export a mbed-cli project to SW4STM32 IDE, the project doesn't include any code.

I've tried opening it from the `.cproj` file or import it with the import option, using the `Existing Project into Workspace` or `Existing Code as Makefile Project`.

The project contains `Includes` folder, `mbed_config.h`, `exporter.yaml`, `GettingStarted.htm` and `Makefile`.

Similar operation to IAR  IDE works fine.

0xc0170, 2017-01-16:
cc @theotherjimmy @bcostm

theotherjimmy, 2017-01-17:
@abouillot Could you try exporting to eclipse_gcc_arm? It should work for you. The SW4STM32 exporter is currently broken, and it's on my todo list to fix it.

abouillot, 2017-01-19:
eclipse_gcc_arm isn't a recognized toolchain by mbed-cli.

theotherjimmy, 2017-01-19:
how about `mbed export -i eclipse_gcc_arm -m ` your target?

abouillot, 2017-01-19:
Here is the output:

```
F:\mbed-cli\mbed-programs\Nucleo_printf_test>mbed --version
1.0.0

F:\mbed-cli\mbed-programs\Nucleo_printf_test>mbed export -i eclipse_gcc_arm -m NUCLEO_L152RE -c
usage: project.py [-h] [-m MCU] [-i IDE] [-c] [-p PROGRAM] [-n PROGRAM] [-b]
                  [-L] [-S] [-E] [--source SOURCE_DIR] [-D MACROS]
project.py: error: argument -i: eclipse_gcc_arm is not a supported toolchain. Supported toolchains are:
atmelstudio,  coide,        ds5_5,        e2studio,     emblocks,
gcc_arm,      iar,          kds,          lpcxpresso,   simplicityv3,
sw4stm32,     uvision,      uvision4,     uvision5
[mbed] ERROR: "python" returned error code 2.
[mbed] ERROR: Command "python -u F:\mbed-cli\mbed-programs\Nucleo_printf_test\.temp\tools\project.py -i eclipse_gcc_arm -m NUCLEO_L152RE -c --source ." in "F:\mbed-cli\mbed-programs\Nucleo_printf_test"
---
```

theotherjimmy, 2017-01-19:
Alright. Is this an mbed 2 project?

theotherjimmy, 2017-01-19:
Or is your mbed-os pretty back-dated?

abouillot, 2017-01-19:
you can find the code @ https://developer.mbed.org/users/abouillot/code/Nucleo_printf_test/ . This project has been created less than a week ago, I don't think any part of it is so outdated.

theotherjimmy, 2017-01-19:
I don't have access to that repo.

abouillot, 2017-01-19:
I change the repo to public.

theotherjimmy, 2017-01-19:
So, mbed-dev uses the tools from here: https://developer.mbed.org/users/mbed_official/code/mbed-sdk-tools/
I have opened a PR to update the tools here: https://developer.mbed.org/users/mbed_official/code/mbed-sdk-tools/pull-request/1

When I get that PR merged, you can then start using the generic makefile-based eclipse exporters.

screamerbg, 2017-04-25:
@theotherjimmy @abouillot Is this still an issue?

theotherjimmy, 2017-04-25:
@abouillot The workaround (using `make_gcc_arm` or `eclipse_gcc_arm`) should now be possible. Could you try them out?

screamerbg, 2017-06-21:
@abouillot @theotherjimmy bump

closed 2017-07-05 by screamerbg