ARMmbed/mbed-cli #421: Projects are exported as Unnamed_Project

abouillot, 2017-01-14:
When exporting to SW4STM or IAR IDEs, the projects are called Unnamed_project. This make working on different projects in parallel complex.

screamerbg, 2017-01-18:
cc @theotherjimmy @sarahmarshy @0xc0170

theotherjimmy, 2017-01-19:
@abouillot I cannot reproduce your problem with the latests version of mbed-os. Could you tell me which version of mbed-os you are using? or if you are using a version of mbed what the release number is?

abouillot, 2017-01-19:
I'm using mbed-cli v1.0.0.

```
F:\mbed-cli\mbed-programs\Nucleo_printf_test>mbed export -i sw4stm32 -m NUCLEO_L152RE -c
Scan: lib
Scan: src
Scan: .

Successful:
  * NUCLEO_L152RE::sw4stm32     .\projectfiles\sw4stm32_NUCLEO_L152RE\Unnamed_Project

F:\mbed-cli\mbed-programs\Nucleo_printf_test>mbed --version
1.0.0
```

theotherjimmy, 2017-01-19:
@abouillot Sorry for the confusion, I was asking for the version of mbed-os, not mbed-cli. Is the program you are using published somewhere?

abouillot, 2017-01-19:
Code can be found @ https://developer.mbed.org/users/abouillot/code/Nucleo_printf_test/ , but there is nothing special about it.

theotherjimmy, 2017-01-19:
So I don't have access to that repo... do you have an `mbed.bld` or a `mbed-os.lib` in the project's root directory?

bridadan, 2017-01-19:
You can also run `mbed ls` and post the output here. That will give a recursive log of all libraries' commit hashes.

abouillot, 2017-01-19:
I have mbed-dev.lib at the root

screamerbg, 2018-05-26:
@adustm Could you please update your Mbed CLI to the latest version (1.7.1) and then run the following commands
```
cd <to your program>
cd mbed
mbed update # this will update the mbed library
cd ..
mbed export -i sw4stm32 -m NUCLEO_L152RE -c
```

Please let me know if that fixes your problem

0xc0170, 2018-06-13:
I tested this locally with 5.8.6, can't reproduce anymore. I'll close this issue as resolved

closed 2018-06-13 by 0xc0170