ARMmbed/mbed-cli #422: mbed remove doesnt work without first mbed deploy

mbartling, 2017-01-17:
I have an mbed2 project I just imported from mercurial that I am porting to mbed OS 5, and when I run the following commands I get an error `mbed not found`

```
mbed target K64F
mbed toolchain GCC_ARM
mbed remove mbed
mbed add mbed-os
mbed deploy
```

However this works

```
mbed target K64F
mbed toolchain GCC_ARM
mbed deploy                        #<=== Moved 
mbed remove mbed
mbed add mbed-os
```

bridadan, 2017-01-17:
@mbartling Can you post the link to the mercurial repo? And can post the exact command you used to import the repo?

screamerbg, 2017-03-08:
@mbartling bump

mbartling, 2017-03-08:
No longer having this issue. Just checked with all of my mbed 2 examples

closed 2017-03-08 by mbartling