ARMmbed/mbed-cli #424: mbed-cli allows using hashless libs

teetak01, 2017-01-19:
I noticed that I can remove the #HASH completely from .lib file. If this is then also pushed to github, mbed deploy will always pull in from head of master.

Just wondering that is this working as designed or possibly a bug in how mbed-cli handles .lib files. Considering relation to this issue: https://github.com/ARMmbed/mbed-cli/issues/224

bridadan, 2017-01-19:
@teetak01 I believe the functionality should be that if there is no #HASH, it would pull from master and then save the .lib with the #HASH from the tip of master. Is this the behavior you're seeing?

teetak01, 2017-01-19:
Yes, that is what I'm observing. mbed-cli was earlier pretty strict about .lib file usage so was wondering correctly. The current functionality is fine for me.

bridadan, 2017-01-19:
> The current functionality is fine for me.

I think so too. I'll close this then, thanks for raising your concern!

closed 2017-01-19 by bridadan