ARMmbed/mbed-cli #426: Mecurial dependency

clivebluston, 2017-01-24:
mbed-cli depends on Mercurial.
Is this really necassary?

I am using Ubuntu 12.03 which only supports an older version of Mercurial than required.

Clive

bridadan, 2017-01-24:
mbed-cli depends on Mercurial because the repositories on developer.mbed.org are mercurial repositories. If you don't use mercurial repositories I think it should work just fine without it installed. Are you seeing something different?

clivebluston, 2017-01-25:
Yes. I am creating a new mbed-os app from scratch. At some point it complains about downloading tools and that I should have hg on my command path.

bridadan, 2017-01-25:
Ah I see. Could you post the verbose log from when you try to create a new mbed-os app from scratch?

And what version of Mercurial are you using? If you add it to your path does that help?

screamerbg, 2017-02-22:
With repository hosting like BitBucket, the repository url can be either mercurial or git repositories. Therefore mbed CLI has to probe each method (git clone and mercurial clone) while importing.

theotherjimmy, 2017-11-28:
Closing, as the question is answered.

closed 2017-11-28 by theotherjimmy