ARMmbed/mbed-cli #427: Keil uVision export change to allow the use of the free Keil uVision STM 32 L0 and F0 license

michaelVisimid, 2017-01-24:
The CLI exporter is working great for Keil exports. I would like to suggest that the field 

<Cpu></Cpu>

in the generated .uvprojx file list the processor type be changed during export. To use the free STM32 M0/0+ license the .uvprojx file must be modified as shown below or the compiler will not work with the free license. It will work with the full pro license in its current state with no issues:

<Cpu></Cpu>CPUTYPE("Cortex-M0+")</Cpu>

I suggest it as the STM32 M0/0+ free Keil license is attractive to new users and this fix might not be immediately obvious to them.

bridadan, 2017-01-30:
@sarahmarshy @theotherjimmy any thoughts on this? The change sounds simple enough and we could verify that nothing breaks by running the CI test on it.

theotherjimmy, 2017-01-30:
Sounds great! @michaelVisimid The generated `.uprojx` file is generated from https://github.com/ARMmbed/mbed-os/blob/master/tools/export/uvision/uvision.tmpl Which is filled by the python https://github.com/ARMmbed/mbed-os/blob/master/tools/export/uvision/__init__.py 

If you would like to do this yourself, I will happily answer questions about how the tools work.

michaelVisimid, 2017-01-30:
If I can catch a free moment I will. Busy as a family of beavers these days. I think the fix to line 19 is as follows:

`<Cpu>CPUTYPE("{{device.core}}")</Cpu>`

theotherjimmy, 2017-01-30:
@michaelVisimid I can make the PR for you with that Info. Just a moment.

michaelVisimid, 2017-01-30:
Cool thanks!

theotherjimmy, 2017-01-30:
There you go https://github.com/ARMmbed/mbed-os/issues/3661

screamerbg, 2017-02-22:
@michaelVisimid please close this if resolved with the PROBLEM @theotherjimmy linked above

closed 2017-02-22 by michaelVisimid