ARMmbed/mbed-cli #429: Error when using 'mbed new .' on exported Online project

MarceloSalazar, 2017-01-29:
Reproduce with these steps:

1. Download ZIP file from existing online project:
https://developer.mbed.org/teams/mbed-os-examples/code/mbed-os-example-blinky/export

2. Unzip and change to directory

3. `dir /A'

```
29/01/2017  15:10                32 .gitignore
29/01/2017  15:10    <DIR>          .hg
29/01/2017  15:10    <DIR>          img
29/01/2017  15:10               213 main.cpp
29/01/2017  15:12    <DIR>          mbed-os
29/01/2017  15:10                77 mbed-os.lib
29/01/2017  15:10             2,737 README.md
```

3. `mbed new .`

Output:
```
[mbed] ERROR: A repository already exists in "C:\mbed\xxx\mbed-os-example-blinky_zip_NUCLEO_F429ZI\mbed-os-example-blinky" based on git. Please select a different name or location.
---
```

Either the exported should not have a `.gitignore` directory, or mbed CLI should be able to handle this.

bridadan, 2017-01-30:
Hey marcelo, I'm not sure if that's quite the correct command to use (though I think the behavior could be modified to be smarter about this).

Can you try `mbed deploy` instead of `mbed new .`?

bridadan, 2017-01-31:
@MarceloSalazar sorry didn't realize the code was already there!

Ok, so pretty sure this is related to https://github.com/ARMmbed/mbed-cli/pull/397.

If you create a `.mbed` file then mbed-cli stops complaining. I think `mbed new .` should probably be smart enough to realize that it's already a project and just add the `.mbed` file. @screamerbg what do you think?

bridadan, 2017-01-31:
Also side note @MarceloSalazar, any reason you're not using `mbed import` instead of the zip exporters?

```
mbed import https://developer.mbed.org/teams/mbed-os-examples/code/mbed-os-example-blinky
```

I agree the exporters should work as well (and mbed CLI should be smart enough to cope with them), but it seems a lot less complicated to just run the command above. You shouldn't have this same problem then.

MarceloSalazar, 2017-02-14:
Using `mbed import` and then `mbed export ...` is a workaround.
It would be nice to fix the `mbed new .` to add the `.mbed` file.

screamerbg, 2017-03-08:
@theotherjimmy Can we make the online exporters to create .mbed file in the root of the program containing `ROOT=.` for when exporting a repository (e.g. .hg/.git inside?)

theotherjimmy, 2017-03-08:
Yes. That's should be a quick change.

screamerbg, 2017-04-25:
@MarceloSalazar @theotherjimmy Any progress on this?

theotherjimmy, 2017-04-25:
Putting up the patch now.

theotherjimmy, 2017-04-25:
https://github.com/ARMmbed/mbed-os/pull/4224

theotherjimmy, 2017-05-08:
Alright: Now we have to update the tools in the website... I'll keep you posted.

screamerbg, 2017-06-21:
@theotherjimmy bump

theotherjimmy, 2017-08-01:
Currently available on the website. Closing.

closed 2017-08-01 by theotherjimmy