ARMmbed/mbed-cli #430: mbed sync ignores forks of mbed-os / sets lib reference incorrectly

maclobdell, 2017-02-01:
While testing an application example using a fork of mbed OS (from a PR), I wanted to sync the mbed-os library to the mbed-os fork instead of the main repo location. 

However, mbed sync seems to always assume that mbed-os is located at https://github.com/armmbed/mbed-os/

Steps to recreate: 

mbed import mbed-os-example-mesh-minimal
cd mbed-os-example-mesh-minimal\mbed-os
git remote add myremote https://github.com/pradeep-gr/mbed-os5-onsemi
git fetch myremote
git checkout myremote/master
cd ..
mbed sync

This results in mbed-os.lib containing an invalid reference to: 
https://github.com/ARMmbed/mbed-os/#079aedf2531e23a504b5f1ca761ec87033792f0f
which does not exist. 
It should be:
https://github.com/pradeep-gr/mbed-os5-onsemi/079aedf2531e23a504b5f1ca761ec87033792f0f

screamerbg, 2018-05-26:
@maclobdell We should add some documentation to point out that Mbed CLI "prefers" `origin` remote and therefore you should change the "origin" URL if you're pulling from a fork. Is this something you could look at with perhaps @AnotherButler ?

maclobdell, 2018-06-05:
ah, I see.  I thought that it could figure out which remote you want to specify based on if you have a specific remote/branch checked out, but I guess that isn't possible.   I agree this would be good to document.  Perhaps also this could be mentioned in the help for mbed sync (mbed sync -h).

closed 2019-10-31 by ciarmcom