ARMmbed/mbed-cli #432: mbed-cli downloads mbed 2.0 tools during library updates

bremoran, 2017-02-02:
When I use mbed-cli to update a single library in my tree for testing/development, it downloads redundant copies of the mbed 2.0 build tools.

mbed-cli should not download the mbed 2.0 build tools when it is inside a mbed OS 5.0 library, which is in a mbed OS 5.0 application.

bridadan, 2017-02-02:
Hi @bremoran, could you provide the following:

1. Your mbed CLI version `mbed --version`
2. The commands we can use to reproduce this issue.

Thanks!

bremoran, 2017-02-02:
```
$ mbed --version
1.0.0
```

Suppose we have an mbed OS 5 application: app. It uses a library: lib. We want to change lib to a new branch: branch.

1. `git clone http://domain.name/path/to/app.git`
2. `cd app`
3. `mbed deploy`
4. `cd lib`
5. `mbed update branch`

mbed will report:
```
[mbed] Couldn't find build tools in your program. Downloading the mbed 2.0 SDK tools...
```

bridadan, 2017-02-02:
@bremoran I wasn't able to reproduce your issue. I'm using mbed CLI 1.0.0 on Windows 7. Here are the commands I ran. Can you reproduce the error on your machine with the following?

```
git clone https://github.com/ARMmbed/mbed-os-example-mesh-minimal
cd mbed-os-example-mesh-minimal
mbed deploy
cd atmel-rf-driver
mbed update retries
```

`retries` was just a random branch I found the atmel-rf-driver repo.

screamerbg, 2017-04-25:
@bremoran Could you provide more details for us to reproduce?

screamerbg, 2017-06-21:
Closing as there is no instructions to reproduce. Please reopen if needed.

closed 2017-06-21 by screamerbg