ARMmbed/mbed-cli #434: Mbed cli export & SW4STM32

fluuffy, 2017-02-07:
I get the following error when I run `mbed  export supported` on my nucleo_303K8:

Traceback (most recent call last):
  File "C:\Users\nj\mbed_blinky\.temp\tools\project.py", line 14, in <module>

    from tools.export import EXPORTERS, mcu_ide_matrix
  File "C:\Users\nj\mbed_blinky\.temp\tools\export\__init__.py", line 18, in
<module>
    from tools.export import codered, ds5_5, iar, makefile
  File "C:\Users\nj\mbed_blinky\.temp\tools\export\iar\__init__.py", line 12,
 in <module>
    from tools.export.cmsis import DeviceCMSIS
  File "C:\Users\nj\mbed_blinky\.temp\tools\export\cmsis\__init__.py", line 9
, in <module>
    from tools.arm_pack_manager import Cache
  File "C:\Users\nj\mbed_blinky\.temp\tools\arm_pack_manager\__init__.py", li
ne 2, in <module>
    from bs4 import BeautifulSoup
ImportError: No module named bs4
[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u C:\Users\nj\mbed_blinky\.temp\tools\project.
py -S" in "C:\Users\nj\mbed_blinky"
---

I get the same error when I try to export to SW4STM32

fluuffy, 2017-02-07:
I don't know if this helps, but here is what I get when I run `mbed detect`

'C:\Users\59317\mbed_blinky>mbed detect

[mbed] Detected NUCLEO_F303K8, port COM9, mounted G:
[mbed] Supported toolchains for NUCLEO_F303K8
+--------+-----------+-----------+-----+---------+-----+
| Target | mbed OS 2 | mbed OS 5 | ARM | GCC_ARM | IAR |
+--------+-----------+-----------+-----+---------+-----+
+--------+-----------+-----------+-----+---------+-----+
Supported targets: 0'

bridadan, 2017-02-07:
Hi @fluuffy,

Hmm, mbed CLI should have installed the requirements for you. You may need to do this manually in this case.

Please run the following commands:

```
cd C:\Users\nj\mbed_blinky\.temp\tools
pip install -r requirements.txt
```

When that's done, `cd` back to your project directory and retry your commands.

fluuffy, 2017-02-13:
This worked for me:
`mbed compile -v -t GCC_ARM -m nucleo_f303k8 --profile mbed-os/tools/profiles/monkey.json`

I couldn't view printf outputs from UART, so I had to mock up monkey.json

the json contains the following settings:
`{
    "GCC_ARM": {
        "common": ["-c", "-Wall", "-Wextra",
                   "-Wno-unused-parameter", "-Wno-missing-field-initializers",
                   "-fmessage-length=0", "-fno-exceptions", "-fno-builtin",
                   "-ffunction-sections", "-fdata-sections", "-funsigned-char",
                   "-MMD", "-fno-delete-null-pointer-checks",
                   "-fomit-frame-pointer", "-O0", "-g"],
        "asm": ["-x", "assembler-with-cpp"],
        "c": ["-std=gnu99"],
        "cxx": ["-std=gnu++98", "-fno-rtti", "-Wvla"],
        "ld": ["-Wl,--gc-sections", "-Wl,--wrap,main", "-Wl,--wrap,_malloc_r",
               "-Wl,--wrap,_free_r", "-Wl,--wrap,_realloc_r",
               "-Wl,--wrap,_calloc_r", "-Wl,--wrap,exit", "-Wl,--wrap,atexit", "-u _printf_float", "-u _scanf_float"]
    },
    "ARM": {
        "common": ["-c", "--gnu", "-Otime", "--split_sections",
                   "--apcs=interwork", "--brief_diagnostics", "--restrict",
                   "--multibyte_chars", "-O3"],
        "asm": [],
        "c": ["--md", "--no_depend_system_headers", "--c99", "-D__ASSERT_MSG"],
        "cxx": ["--cpp", "--no_rtti", "--no_vla"],
        "ld": []
    },
    "uARM": {
        "common": ["-c", "--gnu", "-Otime", "--split_sections",
                   "--apcs=interwork", "--brief_diagnostics", "--restrict",
                   "--multibyte_chars", "-O3", "-D__MICROLIB",
                   "--library_type=microlib", "-DMBED_RTOS_SINGLE_THREAD"],
        "asm": [],
        "c": ["--md", "--no_depend_system_headers", "--c99", "-D__ASSERT_MSG"],
        "cxx": ["--cpp", "--no_rtti", "--no_vla"],
        "ld": ["--library_type=microlib"]
    },
    "IAR": {
        "common": [
            "--no_wrap_diagnostics", "-e",
            "--diag_suppress=Pa050,Pa084,Pa093,Pa082", "-Oh"],
        "asm": [],
        "c": ["--vla"],
        "cxx": ["--guard_calls", "--no_static_destruction"],
        "ld": ["--skip_dynamic_initialization", "--threaded_lib"]
    }
}`

closed 2017-02-13 by fluuffy