ARMmbed/mbed-cli #435: mbed cli gives error and mbed online compiler run without error on same code

masoudr, 2017-02-07:
Hi,
I know that my question maybe not relevant but I have a problem with mbed-cli and not with mbed online compiler.
I can compile my code with no problem on mbed online compiler, but when I want to compile it with mbed-cli and GCC_ARM I get this error log. Is there any fix for this problem or make mbed-cli compiler like the online compiler that can handle these errors? Should I use another compiler instead of GCC_ARM?
thanks

> Compile [100.0%]: main.cpp
> [Error] simple-mbed-client.h@340,10: 'void SimpleMbedClientBase::keep_alive()' is private
> [Error] main.cpp@23,27: within this context
> [Error] main.cpp@25,67: cannot pass objects of non-trivially-copyable type 'std::string {aka class std::basic_string<char>}' through '...'
> [Warning] main.cpp@24,22: unused variable 'ledResource' [-Wunused-variable]
> [ERROR] In file included from .\source\main.cpp:4:0:
> ./simple-mbed-client/simple-mbed-client.h: In function 'int main()':
> ./simple-mbed-client/simple-mbed-client.h:340:10: error: 'void SimpleMbedClientBase::keep_alive()' is private
>      void keep_alive() {
>           ^
> .\source\main.cpp:23:27: error: within this context
>          client.keep_alive();
>                            ^
> .\source\main.cpp:25:67: error: cannot pass objects of non-trivially-copyable type 'std::string {aka class std::basic_string<char>}' through '...'
>         printf("Value is currently %s\n", static_cast<string>(name));
>                                                                    ^
> .\source\main.cpp:24:22: warning: unused variable 'ledResource' [-Wunused-variable]
>          M2MResource* ledResource = name.get_resource();
>                       ^
> 
> [mbed] ERROR: "python" returned error code 1.
> [mbed] ERROR: Command "python -u C:\Users\Masoud\Desktop\Work\mbed\simple-mbed-client-example-F746-me\mbed-os\tools\make.py -t GCC_ARM -m NUCLEO_F746ZG --source . --build .\BUILD\NUCLEO_F746ZG\GCC_ARM" in "C:\Users\Masoud\Desktop\Work\mbed\simple-mbed-client-example-F746-me"
> ---

bridadan, 2017-02-07:
Hi @masoudr, in the future this should really be raised against the example program you're working from.

I wasn't able to reproduce the issue locally (I'm assuming you're working from [this example](https://developer.mbed.org/teams/sandbox/code/simple-mbed-client-example/) and that you've made the necessary config changes to get it to compile). I'm using GCC_ARM 4.9.3. What version are you using?

masoudr, 2017-02-08:
Thank you for the response. You are right I'm using simple mbed client example. The sample works fine. I created my own program with this library but it gives me errors. I am using `ARM GCC 4.9 2015q3`. For example consider this code:

> #include "mbed.h"
#include "security.h"
#include "easy-connect.h"
#include "simple-mbed-client.h"
#include "rtos.h"
#include "stdio.h"
#include <cstdlib>
int a=1;
string last_value;
SimpleMbedClient client;
DigitalOut led(LED2, 0);
DigitalOut led3(LED3, 0);
void led3_thread(void const *args) {
    while (true) {
        led3 = !led3;
        Thread::wait(a);
    }
}
void registered() {
    led = 1;
}
void name_updated(string new_value) {
    printf("Value is now %s\n", new_value.c_str());
    last_value=new_value.c_str();
}
SimpleResourceString name = client.define_resource("device/0/name", "500", M2MBase::GET_PUT_ALLOWED, true, &name_updated);
int main() {
    Thread thread(led3_thread);
    NetworkInterface* network = easy_connect(true);
    client.setup(network); // returns a bool, check if it's true
    client.on_registered(&registered);
    while (1) {
      wait_ms(10000);
      client.keep_alive();
      M2MResource* ledResource = name.get_resource();
      printf("Value of name is currently %s\n", static_cast<string>(name));
      a=atoi(last_value.c_str());
      printf("Value of a is currently %d\n", a);
    }
}

I can compile it without any errors on mbed online compiler (only with some warnings) but when I use mbed CLI it gives me the following error:

> Compile [100.0%]: main.cpp
[Warning] main.cpp@32,30: 'rtos::Thread::Thread(void (*)(const void*), void*, osPriority, uint32_t, unsigned char*)' is deprecated (declared at ./mbed-os/rtos/Thread.h:189): Thread-spawning constructors hide errors. Replaced by thread.start(callback(task, argument)). [since mbed-os-5.1] [-Wdeprecated-declarations]
[Error] simple-mbed-client.h@340,10: 'void SimpleMbedClientBase::keep_alive()' is private
[Error] main.cpp@38,27: within this context
[Error] main.cpp@40,74: cannot pass objects of non-trivially-copyable type 'std::string {aka class std::basic_string<char>}' through '...'
[Warning] main.cpp@39,21: unused variable 'ledResource' [-Wunused-variable]
[ERROR] .\source\main.cpp: In function 'int main()':
.\source\main.cpp:32:30: warning: 'rtos::Thread::Thread(void (*)(const void*), void*, osPriority, uint32_t, unsigned char*)' is deprecated (declared at ./mbed-os/rtos/Thread.h:189): Thread-spawning constructors hide errors. Replaced by thread.start(callback(task, argument)). [since mbed-os-5.1] [-Wdeprecated-declarations]
     Thread thread(led3_thread);
                              ^
In file included from .\source\main.cpp:4:0:
./simple-mbed-client/simple-mbed-client.h:340:10: error: 'void SimpleMbedClientBase::keep_alive()' is private
     void keep_alive() {
          ^
.\source\main.cpp:38:27: error: within this context
         client.keep_alive();
                           ^
.\source\main.cpp:40:74: error: cannot pass objects of non-trivially-copyable type 'std::string {aka class std::basic_string<char>}' through '...'
       printf("Value of name is currently %s\n", static_cast<string>(name));
                                                                          ^
.\source\main.cpp:39:21: warning: unused variable 'ledResource' [-Wunused-variable]
        M2MResource* ledResource = name.get_resource();
                     ^

[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u C:\Users\Rahimi\Desktop\mbed\CLI\simple-mbed-client-example-F746-me\mbed-os\tools\make.py -t GCC_ARM -m NUCLEO_F746ZG --source . --build .\BUILD\NUCLEO_F746ZG\GCC_ARM" in "C:\Users\Rahimi\Desktop\mbed\CLI\simple-mbed-client-example-F746-me"
---

bridadan, 2017-02-08:
I see, I'm afraid I can't think of anything off the top of my head to fix this. These are compiler errors, so not an issue with mbed CLI.

The online compiler uses the ARM compiler (bundled with Keil uVision or available commercially and standalone [here](https://developer.arm.com/products/software-development-tools/compilers/arm-compiler-5/downloads)), so you could always try using that locally.

masoudr, 2017-02-08:
Thank you.
I download it and installed it. when I want to compile it gives me errors about not finding the licence. I think this compiler is not free, am I right?

bridadan, 2017-02-08:
> I think this compiler is not free, am I right?

That's correct

masoudr, 2017-02-08:
thank you. I hope maybe the ARM release a new free compiler like its online compiler to save these issues.

masoudr, 2017-02-08:
Finally, I solved the problem. the problem was with `client.keep_alive();`. I removed it and it worked.
I hope it helps.

bridadan, 2017-02-08:
Ok, thanks for the update @masoudr, I'll go ahead and close this.

closed 2017-02-08 by bridadan