ARMmbed/mbed-cli #436: Creating a new program for mbed OS 2, the mbed library isn't downloaded

eavelardev, 2017-02-15:
macOS Sierra
mbed 1.0.0

When I create a new mbed OS 2 program the mbed library isn't downloaded automatically. I execute `mbed update` inside the project folder to resolve.

screamerbg, 2017-02-22:
@eavelardev What command do you use?

eavelardev, 2017-02-22:
I use `mbed new my_project_name --mbedlib`

screamerbg, 2017-02-28:
Hi,

I just used `mbed new <program> --mbedlib` and worked fine. Here is the output

```
D:\1>mbed --version
1.0.0

D:\1>mbed new test-classic --mbedlib
[mbed] Creating new program "test-classic" (git)
[mbed] Adding library "mbed" from "https://mbed.org/users/mbed_official/code/mbed/builds" at latest revision in the current branch
[mbed] Downloading library build "ef9c61f8c49f" (might take a minute)
[mbed] Unpacking library build "ef9c61f8c49f" in "D:\1\test-classic\mbed"
[mbed] Updating reference "mbed" -> "https://mbed.org/users/mbed_official/code/mbed/builds/ef9c61f8c49f"
[mbed] Couldn't find build tools in your program. Downloading the mbed 2.0 SDK tools...
```

screamerbg, 2017-03-08:
@eavelardev bump

screamerbg, 2017-06-21:
Closing as we're not able to reproduce. Please reopen if needed.

michaelVisimid, 2018-02-27:
The issue still exists with version 1.4. This is pretty easily reproduced and still an issue. After you changed to your test program's directory you should have either run tree to see that the mbed libraries are not downloaded, or added a main.cpp which would have showed you the exact same thing. I'm not quite sure how the recreate steps taken would have recreated the issue as at no point were the libraries actually checked for. Below is the output from the command prompt where I should that the issue still exists and that adding main.cpp and importing mbed-dev allows the program to compile. For that matter, why doesn't the CLI create a bare minimum main.cpp that looks something like this:

`#include "mbed.h"

int main() {
    while(1) {      
    }
}

`
Here is the command prompt output which is easily recreated with the use of either tree or attempting to compile with main.cpp. 

`C:\Users\mbaker\Desktop\test>py -2.7 -m mbed --version
1.4.0

C:\Users\mbaker\Desktop\test>py -2.7 -m mbed new test-classic --mbedlib
[mbed] Creating new program "test-classic" (git)
[mbed] Adding library "mbed" from "https://mbed.org/users/mbed_official/code/mbed/builds" at latest revision in the current branch
[mbed] Updating reference "mbed" -> "https://mbed.org/users/mbed_official/code/mbed/builds/tip"
warning: LF will be replaced by CRLF in mbed.bld.
The file will have its original line endings in your working directory.
[mbed] Couldn't find build tools in your program. Downloading the mbed 2.0 SDK tools...

C:\Users\mbaker\Desktop\test>cd test-classic

C:\Users\mbaker\Desktop\test\test-classic>tree
Folder PATH listing
Volume serial number is FA05-9E87
C:.
├───.temp
│   └───tools
│       ├───.hg
│       │   ├───cache
│       │   └───store
│       │       └───data
│       │           ├───arm__pack__manager
│       │           ├───bootloaders
│       │           │   ├───_m_t_s___d_r_a_g_o_n_f_l_y___f411_r_e
│       │           │   ├───_m_t_s___m_d_o_t___f411_r_e
│       │           │   └───_r_e_a_l_t_e_k___r_t_l8195_a_m
│       │           ├───buildbot
│       │           ├───ci__templates
│       │           │   ├───library__build
│       │           │   └───tests__build
│       │           ├───compliance
│       │           ├───config
│       │           ├───data
│       │           │   └───rpc
│       │           ├───dev
│       │           ├───export
│       │           │   ├───atmelstudio
│       │           │   ├───cdt
│       │           │   ├───cmsis
│       │           │   ├───codered
│       │           │   ├───coide
│       │           │   ├───ds5__5
│       │           │   ├───e2studio
│       │           │   ├───embitz
│       │           │   ├───gnuarmeclipse
│       │           │   ├───iar
│       │           │   ├───kds
│       │           │   ├───lpcxpresso
│       │           │   ├───makefile
│       │           │   ├───mcuxpresso
│       │           │   ├───qtcreator
│       │           │   ├───simplicity
│       │           │   ├───sw4stm32
│       │           │   ├───uvision
│       │           │   ├───vscode
│       │           │   └───zip
[... 358 lines cut ...]
| mbed-dev\targets\TARGET_STM\TARGET_STM32L0\TARGET_NUCLEO_L073RZ\device\system_clock.o                          |   432 |     0 |    0 |
| mbed-dev\targets\TARGET_STM\TARGET_STM32L0\device\stm32l0xx_hal.o                                              |    62 |     0 |    4 |
| mbed-dev\targets\TARGET_STM\TARGET_STM32L0\device\stm32l0xx_hal_rcc.o                                          |  1824 |     0 |    0 |
| mbed-dev\targets\TARGET_STM\TARGET_STM32L0\device\stm32l0xx_hal_rcc_ex.o                                       |   416 |     0 |    0 |
| mbed-dev\targets\TARGET_STM\TARGET_STM32L0\device\stm32l0xx_hal_tim.o                                          |   270 |     0 |    0 |
| mbed-dev\targets\TARGET_STM\TARGET_STM32L0\device\stm32l0xx_hal_uart.o                                         |  1159 |     0 |    0 |
| mbed-dev\targets\TARGET_STM\TARGET_STM32L0\device\system_stm32l0xx.o                                           |   173 |     4 |    0 |
| mbed-dev\targets\TARGET_STM\TARGET_STM32L0\serial_device.o                                                     |    72 |     0 |  560 |
| mbed-dev\targets\TARGET_STM\gpio_api.o                                                                         |   439 |     0 |    0 |
| mbed-dev\targets\TARGET_STM\hal_tick_16b.o                                                                     |   328 |     0 |    4 |
| mbed-dev\targets\TARGET_STM\mbed_overrides.o                                                                   |    20 |     0 |    0 |
| mbed-dev\targets\TARGET_STM\pinmap.o                                                                           |   509 |     0 |    0 |
| mbed-dev\targets\TARGET_STM\serial_api.o                                                                       |   952 |     0 |   80 |
| mbed-dev\targets\TARGET_STM\us_ticker_16b.o                                                                    |   202 |     0 |   64 |
| Subtotals                                                                                                      | 45602 |  2504 | 1272 |
+----------------------------------------------------------------------------------------------------------------+-------+-------+------+
Total Static RAM memory (data + bss): 3776 bytes
Total Flash memory (text + data): 48106 bytes

Image: .\BUILD\NUCLEO_L073RZ\GCC_ARM\test-classic.bin

C:\Users\mbaker\Desktop\test\test-classic>tree
`

I have omitted the final tree output because it is quite long since importing mbed-dev imports the configuration for every mbed board.

theotherjimmy, 2018-02-27:
@screamerbg I think the comment above provides ample steps to reproduce to open this issue.

michaelVisimid, 2018-02-27:
mbed-os programs have the same bug.

screamerbg, 2018-05-26:
@michaelVisimid @eavelardev I managed to reproduce the issue. Fix is on its way.

screamerbg, 2018-05-26:
@eavelardev @michaelVisimid PR #680 is now available which address this issue. Could you please try it and let me know if that fixes it for you as well?

screamerbg, 2018-06-18:
This is now tracked in #698

closed 2018-06-18 by screamerbg