ARMmbed/mbed-cli #439: Please add a define with the project/module commit ID to the build

bremoran, 2017-02-22:
It is useful to know exactly which revision of firmware a device is running, particularly for examining bug reports. If mbed-cli provided a define for the application commit ID, that would make it easy for application developers to add this information to their traces. If the module commit ID were available for each module, that would also be very useful, for similar reasons.

cc @screamerbg @sg-

screamerbg, 2017-02-22:
@bremoran thanks for idea. To confirm, you are looking for a defined macro like MBED_APP_COMMIT="edc7e83d3ea81cab63fe1e7e5dcfd022ca67a21a", right?

bremoran, 2017-02-22:
Yes, that would be great. It would also be good to have something like:
`MBED_MODULE_COMMIT` which is defined per module, but I don't know how doable that is. At least with the app-level commit ID, it should be possible to drill down to the individual module commit IDs from the outside.

screamerbg, 2017-06-21:
This was introduced in https://github.com/ARMmbed/mbed-cli/issues/449, but was not approved. Closing for now. Please reopen if you feel that further discussion is needed.

closed 2017-06-21 by screamerbg