ARMmbed/mbed-cli #440: mbed update has unexpected behaviour with caching enabled

bremoran, 2017-02-22:
I have some config set:
```
$ mbed config -L
[mbed] Global config:
CACHE=/var/mbed
PROTOCOL=ssh
```

If I pick a project I have used before, that has had numerous commits since the last time I used it, and do `mbed import`, mbed correctly imports from the cache.

This means that when I do `mbed update master`, mbed does fetch the remotes correctly, but it doesn't update the local branches, so master is the master that was in use when the cache was populated.

If caching were turned off, this behaviour would be completely different, fetching the tip of master, from when I did the import.

screamerbg, 2017-03-23:
@bremoran Thanks for the bug report. Can you please try PR #462?

screamerbg, 2017-03-28:
@bremoran Bump :)

screamerbg, 2017-04-25:
@bremoran could you please try to reproduce with mbed CLI v1.1 (latest) ?

closed 2017-06-21 by screamerbg