ARMmbed/mbed-cli #442: Compile tests fails due to multiple definitions of 'main'

pd0wm, 2017-02-25:
When running test on a blink demo project which also has a main.cpp all the tests fail to build due to multiple definitions of 'main'.

How do you prevent the test runner to including regular source files?

screamerbg, 2017-02-28:
@pd0wm Currently the only way to exclude the source files from the test runner is either to:
* move the offending file outside of the codebase
* use .mbedgignore file to ignore the offending file

More information on .mbedignore can be found here https://docs.mbed.com/docs/mbedmicro-api/en/latest/ignoring_files_from_build/

pd0wm, 2017-02-28:
What would be the correct way to fix this problem in mbed-cli? Maybe provide an optional setting for your main.cpp in your .mbed file. When running tests this file could be excluded.

screamerbg, 2017-03-08:
@pd0wm there are projects that don't use main.cpp for the main() routine, but rather different file naming. Having a setting that controls whether you have main.cpp file or not would be rather confusing.

This could be solved differently and more intelligently, e.g.
* a pre-main() step in the mbed OS startup code that allows tests to hook early
* linker step that detects the location of main() and doesn't link that file (not great as main.cpp might define/include other things)

Adding few more mbed folks on this: @sg- @geky @theotherjimmy @bridadan

geky, 2017-03-08:
I think the correct solution is to decide the entry point in the linking step. However this would require a different name for main in the tests.

I believe @c1728p9 and @theotherjimmy are looking at this problem is a more general context?

theotherjimmy, 2017-03-08:
@geky see https://github.com/ARMmbed/mbed-os/pull/3730 for the API available for this in the toolchains.

c1728p9, 2017-03-09:
I have a PR up specifically to support the redirection of main - https://github.com/ARMmbed/mbed-os/pull/3801

Looping in @bridadan since he had some ideas on this as well.

bridadan, 2017-03-09:
I think the route @theotherjimmy and @c1728p9 are proposing seems the easiest route to fit into the "mbed way". Specifying your application's entry point in metadata is pretty unlike most of the mbed mechanisms, so I think setting this convention would be good enough to keep us moving in the right direction.

bridadan, 2019-03-04:
In release 1.9.0, we added the feature to include the macro `MBED_TEST_MODE` when building with `mbed test` (added by #835). You can then wrap your main application's `main` function with the following to prevent the multiple main function error:

```c++
#if !MBED_TEST_MODE
int main() {
// ...
}
#endif
```

closed 2019-03-04 by bridadan