ARMmbed/mbed-cli #443: a wrong path is set by windows installer 0.33

soramame21, 2017-02-28:
I uninstalled following software manually, and cleaned environmental variable "path". 
python, mercurial, git, mbed-cli, Serial drivers
then downloaded windows installer 0.33 and ran.
https://mbed-media.mbed.com/filer_public/0d/2a/0d2ae854-a9d8-4bfa-b23b-f6ca89394333/mbed_installer_v033_cli_v100.exe

the installer add **"C:\mbed-cli\python\;C:\mbed-cli\python\Scripts;"** into environmental variable "path".
but the mbed.exe/mbed-cli.exe are located under **"C:\mbed-cli\workspace\Scripts"**.
therefore when I type mbed at Command Promt, I got following error.
**'mbed' is not recognized as an internal or external command**

arekzaluski, 2017-02-28:
I'll investigate it.

arekzaluski, 2017-02-28:
@soramame21 I'm not able to reproduce this issue. Can you provide me additional information about your setup? Which Windows version do you use? 32 or 64 bit?
How are you running it? Do you use windows command prompt or "Run mbed CLI" ? Keep in mind that mbed won't be accessible from windows cmd. You should use "Run mbed CLI.bat".

soramame21, 2017-02-28:
my windows 8.1 is 64bit. I just download above file (mbed_installer_v033_cli_v100.exe) and double click it. I didn't use command prompt.

soramame21, 2017-02-28:
for running mbed, I started a command prompt and type '''mbed'''.

arekzaluski, 2017-02-28:
Can you please try "Run mbed CLI" ? It should be on your desktop. We wanted to keep it separate so it is not accessible from command prompt. I'm considering adding it also to the path so it will be available from command prompt.

soramame21, 2017-02-28:
where is the "Run mbed CLI"? is it a link?

arekzaluski, 2017-02-28:
It is a batch file. It should be added to your desktop after successful installation.

soramame21, 2017-02-28:
I got it, it's working!! but not from a regular command prompt.

arekzaluski, 2017-02-28:
Yes, It is done like that by design. I'm considering changing it.

soramame21, 2017-02-28:
then the test is passed.

MarceloSalazar, 2017-02-28:
mbed CLI should be working from the standard command prompt.
Consider this as must-have requirement for a public release

soramame21, 2017-02-28:
I agree, we just update environment variable "path" then it can work for standard command prompt.

arekzaluski, 2017-02-28:
ok, I'll update it as soon as possible.

soramame21, 2017-02-28:
Hi Arek,

This issue (test 1 of OOB v5.4) will be repeated on test 2 to test 4. 
should I go through the test 2 to 4 of OOB v5.4?

MarceloSalazar, 2017-02-28:
I hit the same problem on Windows 10 64-bits

MarceloSalazar, 2017-02-28:
> This issue (test 1 of OOB v5.4) will be repeated on test 2 to test 4.
> should I go through the test 2 to 4 of OOB v5.4?

Yes, I'd suggest you go through other cases ignoring this particular issue (for now).

arekzaluski, 2017-02-28:
I investigated it further and I'm not sure that we want to change it. Currently after installation mbed-cli is encapsulated in Virtual python Environment. Which means that it is separated from user's system. The batch file is starting a Virtual Environment and gives user access to mbed. With this solution we allow user to have a separated instance of mbed-cli already installed and we do not change user's current setup. Information about python Virtual Environment can be found in https://github.com/ARMmbed/mbed-cli#installing-mbed-cli.

BlackstoneEngineering, 2017-03-01:
For reference, the reason we did not add to the global path is because we did not want to pollute the users global path. The intial design for the installer followed the same design template that GCC did. This is also how the Yotta cli tool was installed, not by default, but as a seperate program. That way the impression was given that it was its own tool.

arekzaluski, 2017-03-01:
@BlackstoneEngineering @screamerbg @sg- What do you think about appending it to the end of the path? In that case after using "mbed" or "mbed-cli" in command prompt user will still use his previously installed mbed-cli version if it exists. If not then a version installed using installer will be used.

sg-, 2017-03-01:
Mixing paths and environments is difficult to support. The entry point had always been and imo should remain a virtual environment wrapped in a bat or similar shell script. 

Anyone can follow the directions to install the individual tools and polute the system if they prefer

BlackstoneEngineering, 2017-03-02:
Agreed. I do not want to mess with the system path on a permenant basis. 

That said, this whole thing will get much easier once mbed CLI is not a python VE but a .exe or .app file. Then I would support adding the folder path to the env path.

screamerbg, 2017-06-21:
Closing as it seems to be fixed in v0.4. Please reopen if reoccurs.

closed 2017-06-21 by screamerbg

BlackstoneEngineering, 2019-10-21:
@arekzaluski

D252874, 2019-10-21:
i found my problem i had python version 3.7.4 and looked in the readme in the installer and found out that you need 3.7.11 or higher to work now i got 3.8.0 and it works for me now