ARMmbed/mbed-cli #446: [OOB_5_4] Windows installer cannot install mbed.py

soramame21, 2017-03-01:
when I run mbed_installer_v033_cli_v100.exe and got following popup. I just want to skip Python installation, but mbed.py cannot be installed.
Is it a expected behavior?

[image]

soramame21, 2017-03-07:
@arekzaluski @MarceloSalazar Could you verify this issue and give your comment?

arekzaluski, 2017-03-07:
What version of python did you have installed? Was it 2.7+ or 3+? It should skip only python installation. Currently this check is checking only if python is installed on the system (no matter what version) and I believe that mbed-cli is compatible only with python 2.7+. I extended this check in new build version of installer.

soramame21, 2017-03-07:
@arekzaluski I pre-installed python 2.7 under C:\Program Files (x86) not the directory C:\Python27.

arekzaluski, 2017-03-07:
Thanks, I'll investigate it.

screamerbg, 2017-04-25:
@soramame21 Can you try with the latest installer please?

soramame21, 2017-06-02:
@screamerbg, Sorry for such delayed reply. the windows installer is verified and works well. I will close this issue.

soramame21, 2017-06-02:
I tested on Windows 10 environment.

closed 2017-06-02 by soramame21