ARMmbed/mbed-cli #448: mbed deploy error: The process cannot access the file because it is being used by another process?

bcostm, 2017-03-06:
I have this error when I do this:
```
$ hg clone https://bcostm@developer.mbed.org/teams/ST/code/Nucleo_blink_led/
$ cd Nucleo_blink_led/
$ mbed deploy
[mbed] Adding library "mbed" from "http://mbed.org/users/mbed_official/code/mbed/builds" at rev #0ab6a29f35bf
[mbed] Downloading library build "0ab6a29f35bf" (might take a minute)
[mbed] ERROR: OS Error: The process cannot access the file because it is being used by another process
```

Any idea ?

bridadan, 2017-03-06:
@bcostm do you get any more info if you run `mbed deploy` with the `-v` flag added?

bcostm, 2017-03-06:
Unfortunately not...

BTW I use mbed-cli v1.0.0

screamerbg, 2017-03-06:
@bcostm can you perhaps disable your anti-virus software? Sometimes they check zip files for viruses and takes a while with few hundreds meg zip file

bcostm, 2017-03-07:
It was a good idea... but same problem with the anti-virus disabled.

betzw, 2017-03-07:
@bcostm I think your local settings of proxies and certificates might be outdated.

screamerbg, 2017-03-08:
@bcostm can you try `mbed deploy -vv` please?

bcostm, 2017-03-09:
Same thing:
```
$ mbed deploy -vv
[mbed] Working path "C:\HG\Nucleo_blink_led" (program)
[mbed] Adding library "mbed" from "http://mbed.org/users/mbed_official/code/mbed/builds" at rev #0ab6a29f35bf
[mbed] Setting url to "http://mbed.org/users/mbed_official/code/mbed/builds/tip" in C:\HG\Nucleo_blink_led\mbed
[mbed] Query "hg id --id http://mbed.org/users/mbed_official/code/mbed -r 0ab6a29f35bf" in C:\HG\Nucleo_blink_led\mbed
0ab6a29f35bf
[mbed] Downloading library build "0ab6a29f35bf" (might take a minute)
[mbed] ERROR: OS Error: The process cannot access the file because it is being used by another process
---
```

screamerbg, 2017-03-22:
@bcostm Downloading mbed lib (not mbed os) happens over HTTP. Which means that there is something in your environment that prevents you to download it (like forced HTTP proxy). Do you think you could try in different environment, like home internet?

screamerbg, 2017-04-25:
@bcostm Any news on this?

bcostm, 2017-05-09:
Tested today with https://developer.mbed.org/teams/Semtech/code/SX1276PingPong/ example and every thing has been downloaded correctly, even the mbed library. I close it.

closed 2017-05-09 by bcostm