ARMmbed/mbed-cli #454: Caps lock support for test -n xxxx-xxxxx-xxxx

BlackstoneEngineering, 2017-03-16:
## Problem
When using the `test -n` command the test name must be input in lower case, even if the folders represented by the test name are all in caps. This has lead to a great deal of confusion for developers. 

### example
CI Test shield test for SPI is located in the folder `TESTS/API/SPI`, so to run this test you would use `test -n tests-api-spi`. I have observed multiple developers try to use `test -n TESTS-API-SPI` because the name of the folders are in caps, so they understand the folder names seperated by dashes, but the lower case conversion almost always throws them.

## Proposal
To make the tool more intuitive I propose accepting either caps lock or lower case and then munging it as appropriate before passing it on. That way the tool becomes more intuitive to use.

bridadan, 2017-03-16:
@BlackstoneEngineering agreed. Can we open this against [greentea](https://github.com/ARMmbed/greentea)?

BlackstoneEngineering, 2017-03-17:
We could, will anyone address this?

bridadan, 2017-03-17:
This should be a very easy change. As in add this line:

```python
opts.test_by_names = opts.test_by_name.lower() if opts.test_by_names else opts.test_by_names
```

Above this line: https://github.com/ARMmbed/greentea/blob/master/mbed_greentea/mbed_greentea_cli.py#L867

BlackstoneEngineering, 2017-03-17:
Raised issue here : https://github.com/ARMmbed/greentea/issues/216

closed 2017-03-17 by bridadan