ARMmbed/mbed-cli #455: mbed-cli cache does not notice that commits have disappeared for origin

teetak01, 2017-03-21:
Defect: Minor
Component: mbed-cli cache

This is quite special case, but we noticed an issue with branches and git-history. I had a locally cached branch, which was force-pushed over in Github. This caused the issue that a lib-file pointing to now non-existant HASH stopped working for anyone who was using the component without caching, but on my machine it continued to work as it was always picking up that hash from local cache. Even if I re-cloned the whole application and redeployed from fresh, it was still picking up the local HASH without realizing that it does not exist in Git anymore. This of course caused some temporary headache until we realized that the cache was causing this. Disabling cache then correctly caused a deploy fail.

Should there be some cleanup/warning for this kind of changes? With development branches the git-history can get rewritten quite often.

screamerbg, 2017-03-23:
Thanks for the bug report @teetak01. Can you please try PR #462?

teetak01, 2017-03-27:
I created a test-branch, pointed a lib to that, deployed component. Then removed it from git and after a while, re-deployed and got error for missing HASH. Looks to be working fine @screamerbg !

+1

screamerbg, 2017-03-28:
Thanks for testing it!

screamerbg, 2017-04-25:
Now fixed in mbed CLI v1.1.x

closed 2017-04-25 by screamerbg