ARMmbed/mbed-cli #456: deploy changes my "origin" address

SeppoTakalo, 2017-03-21:
This is hard to replicate, but sometimes it happens.

* I clone mbed OS application directly from its Git repository.
* I call `mbed deploy --protocol ssh`
* When I notice error `warning: no common commits` I press CTRL+C, no need to continue anymore.
* I notice that my `origin` address is changes to point to mbed OS Git repo.

Log
```
septak01@e105190:~/src$ mkdir tmp
septak01@e105190:~/src$ cd tmp/
septak01@e105190:~/src/tmp$ git clone git@github.com:ARMmbed/mbed-os-example-mesh-minimal.git
Cloning into 'mbed-os-example-mesh-minimal'...
remote: Counting objects: 538, done.
remote: Compressing objects: 100% (2/2), done.
remote: Total 538 (delta 0), reused 0 (delta 0), pack-reused 536
Receiving objects: 100% (538/538), 88.30 KiB | 0 bytes/s, done.
Resolving deltas: 100% (267/267), done.
septak01@e105190:~/src/tmp$ cd mbed-os-example-mesh-minimal/
septak01@e105190:~/src/tmp/mbed-os-example-mesh-minimal$ mbed deploy --protocol ssh
[mbed] Adding library "atmel-rf-driver" from "ssh://git@github.com/ARMmbed/atmel-rf-driver.git" at rev #57f22763f4d3
[mbed] Adding library "mbed-os" from "ssh://git@github.com/ARMmbed/mbed-os.git" at rev #f4864dc6429e
warning: no common commits
^C[mbed] User aborted!
Killed by signal 2.
fatal: index-pack failed
septak01@e105190:~/src/tmp/mbed-os-example-mesh-minimal$ git remote -v
origin	ssh://git@github.com/ARMmbed/mbed-os.git (fetch)
origin	ssh://git@github.com/ARMmbed/mbed-os.git (push)
```

SeppoTakalo, 2017-03-21:
@screamerbg Please check.

SeppoTakalo, 2017-03-21:
Happens in my Mac, does not happen in my Linux machine.
Seems that even cloning order is different. In Linux, the `mbed-os.lib` is cloned first. In Mac the `atmel-rf-driver.lib` is cloned first.
Not sure if this is the reason.

screamerbg, 2017-03-23:
@SeppoTakalo I'm unable to reproduce this on my Mac.

```
[18:08:38 mihsto01@e108173-mac:~]$ cd /tmp
[18:08:38 mihsto01@e108173-mac:/tmp]$ mkdir 1
[18:08:39 mihsto01@e108173-mac:/tmp]$ cd 1
[18:08:40 mihsto01@e108173-mac:/tmp/1]$ git clone git@github.com:ARMmbed/mbed-os-example-mesh-minimal.git
Cloning into 'mbed-os-example-mesh-minimal'...
remote: Counting objects: 557, done.
remote: Compressing objects: 100% (21/21), done.
Receiving objects: 100% (557/557), 93.31 KiB | 0 bytes/s, done.
remote: Total 557 (delta 10), reused 0 (delta 0), pack-reused 536
Resolving deltas: 100% (277/277), done.
[18:08:44 mihsto01@e108173-mac:/tmp/1]$ cd mbed-os-example-mesh-minimal/
[18:08:52 mihsto01@e108173-mac:/tmp/1/mbed-os-example-mesh-minimal]$ mbed deploy --protocol ssh
[mbed] Adding library "atmel-rf-driver" from "ssh://git@github.com/ARMmbed/atmel-rf-driver.git" at rev #57f22763f4d3
[mbed] Adding library "mbed-os" from "ssh://git@github.com/ARMmbed/mbed-os.git" at rev #f4864dc6429e
[mbed] Adding library "mcr20a-rf-driver" from "ssh://git@github.com/ARMmbed/mcr20a-rf-driver.git" at rev #d8810e105d7d
[18:09:47 mihsto01@e108173-mac:/tmp/1/mbed-os-example-mesh-minimal]$ git remote -v
origin	git@github.com:ARMmbed/mbed-os-example-mesh-minimal.git (fetch)
origin	git@github.com:ARMmbed/mbed-os-example-mesh-minimal.git (push)
```

Any idea how to reproduce? Can you `cat ~/.mbed/.mbed` ?

SeppoTakalo, 2017-03-27:
Happened to Nanostack border router also.
```
nanostack-border-router$ mbed deploy --protocol ssh
[mbed] Adding library "mbed-os" from "ssh://git@github.com/ARMmbed/mbed-os.git" at rev #305f5c491e34
warning: no common commits
[mbed] Adding library "drivers/atmel-rf-driver" from "ssh://git@github.com/ARMmbed/atmel-rf-driver.git" at rev #57f22763f4d3
warning: no common commits
[mbed] Adding library "drivers/mcr20a-rf-driver" from "ssh://git@github.com/ARMmbed/mcr20a-rf-driver.git" at rev #d8810e105d7d
warning: no common commits
[mbed] Adding library "drivers/sal-stack-nanostack-slip" from "ssh://git@github.com/ARMmbed/sal-stack-nanostack-slip.git" at rev #4a600894da85
warning: no common commits
[mbed] Adding library "drivers/stm-spirit1-rf-driver" from "ssh://git@github.com/ARMmbed/stm-spirit1-rf-driver.git" at rev #bbde15ee85d0
warning: no common commits
[mbed] Adding library "drivers/TARGET_MCUXpresso_MCUS/sal-nanostack-driver-k64f-eth" from "ssh://git@github.com/ARMmbed/sal-nanostack-driver-k64f-eth.git" at rev #ee7e8a9a2122
warning: no common commits
[mbed] Adding library "drivers/TARGET_NUCLEO_F429ZI/sal-nanostack-driver-stm32-eth" from "ssh://git@github.com/ARMmbed/sal-nanostack-driver-stm32-eth.git" at rev #1369c4a6a454
warning: no common commits
septak01@e104897:~/src/tmp/nanostack-border-router$ git remote -v
origin	ssh://git@github.com/ARMmbed/sal-nanostack-driver-stm32-eth.git (fetch)
origin	ssh://git@github.com/ARMmbed/sal-nanostack-driver-stm32-eth.git (push)
```

Seems that it was still able to import it completely.
I really don't know how to reproduce this, as it doesn't always happen. I wonder what combination of settings might cause this.
```
$ cat ~/.mbed/.mbed
CACHE=on
```

```
$ git --version
git version 2.11.1

$ mbed --version
1.0.0
```

SeppoTakalo, 2017-03-27:
I turned the `CACHE=on` to `CACHE=off` and this doesn't happen anymore.

```
nanostack-border-router$ mbed deploy --protocol ssh
[mbed] Adding library "mbed-os" from "ssh://git@github.com/ARMmbed/mbed-os.git" at rev #305f5c491e34
[mbed] Adding library "drivers/atmel-rf-driver" from "ssh://git@github.com/ARMmbed/atmel-rf-driver.git" at rev #57f22763f4d3
[mbed] Adding library "drivers/mcr20a-rf-driver" from "ssh://git@github.com/ARMmbed/mcr20a-rf-driver.git" at rev #d8810e105d7d
[mbed] Adding library "drivers/sal-stack-nanostack-slip" from "ssh://git@github.com/ARMmbed/sal-stack-nanostack-slip.git" at rev #4a600894da85
[mbed] Adding library "drivers/stm-spirit1-rf-driver" from "ssh://git@github.com/ARMmbed/stm-spirit1-rf-driver.git" at rev #bbde15ee85d0
[mbed] Adding library "drivers/TARGET_MCUXpresso_MCUS/sal-nanostack-driver-k64f-eth" from "ssh://git@github.com/ARMmbed/sal-nanostack-driver-k64f-eth.git" at rev #ee7e8a9a2122
[mbed] Adding library "drivers/TARGET_NUCLEO_F429ZI/sal-nanostack-driver-stm32-eth" from "ssh://git@github.com/ARMmbed/sal-nanostack-driver-stm32-eth.git" at rev #1369c4a6a454
```

screamerbg, 2018-05-26:
I think that this was related to the experimental (back then) mbed CLI caching feature. I suppose this is not longer an issue with Mbed CLI 1.7.1 (latest). Closing as there was no activity on this thread for a while, but feel free to reopen if the issue still exists.

closed 2018-05-26 by screamerbg