ARMmbed/mbed-cli #458: 'mbed import' doesn't have a consistent behaviour across multiple executions

MarceloSalazar, 2017-03-22:
This is considered a bug

mbed CLI: 1.0.0
Windows 10
Python: 2.7.13

Summary:
1. Import blinky (I get mbed OS 5.4.1)
2. Update blinky project to mbed OS 5.3.6
3. Move to other directory and import blinky2 (I get mbed mbed OS 5.3.6, whether I should get 5.4.1 - latest!).

Logs:
```
C:\t>mbed import mbed-os-example-blinky
[mbed] Importing program "mbed-os-example-blinky" from "https://github.com/ARMmbed/mbed-os-example-blinky" at latest revision in the current branch
[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at rev #f4864dc6429e

C:\t>cd mbed-os-example-blinky

C:\t\mbed-os-example-blinky>mbed ls -a
mbed-os-example-blinky (https://github.com/ARMmbed/mbed-os-example-blinky#9dbaf52e7d2a)
`- mbed-os (https://github.com/ARMmbed/mbed-os#f4864dc6429e)

C:\t\mbed-os-example-blinky>mbed update mbed-os-5.3.6     # manually switch to old version
[mbed] Updating program "mbed-os-example-blinky" to branch mbed-os-5.3.6
[mbed] Updating library "mbed-os" to rev #5faf4b26c595

C:\t\mbed-os-example-blinky>mbed ls -a
mbed-os-example-blinky (https://github.com/ARMmbed/mbed-os-example-blinky#4d23fd49c818)
`- mbed-os (https://github.com/ARMmbed/mbed-os#5faf4b26c595)

C:\t\mbed-os-example-blinky>cd ..

C:\t>mbed import mbed-os-example-blinky blinky2           # import new project
[mbed] Importing program "blinky2" from "mbed-os-example-blinky" at latest revision in the current branch
Note: checking out '4d23fd49c8182620950e7bfe6edd6a9b58329513'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at rev #5faf4b26c595

C:\t>cd blinky2

C:\t\blinky2>mbed ls -a
blinky2 (C:/t/mbed-os-example-blinky#4d23fd49c818)
`- mbed-os (https://github.com/ARMmbed/mbed-os#5faf4b26c595)    # I get mbed OS 5.3.6 !!!

```

@theotherjimmy

screamerbg, 2017-03-23:
@MarceloSalazar 
mbed CLI (and also Git and Mercurial) support importing from local directories. E.g. `git clone <local git repo folder> <dest folder>` is a valid operation.

When you first import `mbed-os-example-blinky`, you don't specify a destination name and therefore mbed CLI uses the repository name - `mbed-os-example-blinky`

When you import the second time, given there is local folder `mbed-os-example-blinky` mbed CLI (and Git/Mercurial) pick that instead and import from the local folder.

Perhaps we should throw a warning that local folder is imported, not URL.

MarceloSalazar, 2017-03-23:
> When you import the second time, given there is local folder mbed-os-example-blinky mbed CLI (and Git/Mercurial) pick that instead and import from the local folder.

IMO, this doesn't make sense from the *user* point of view.
mbed CLI should use the name that is explicitly specified as dest folder (regardless on whether there is an existing folder or not) and import the latest changes from master. Otherwise there is no reproducibility on executing the import command multiple times.

MarceloSalazar, 2017-03-23:
It seems this is a misunderstanding related to the possibility of importing an example without using the github url. We should probably reinforce the usage of the url in our example's docs (it's not the scope for this ticket).
The doc here clearly shows different alternatives to import a program:
https://github.com/ARMmbed/mbed-cli#importing-an-existing-program

I understand the possibility of importing a local folder (already supported by git and mercurial), which is a useful feature and a valid behaviour, as explained by @screamerbg
As discussed we could:

- Add a line into the docs explaining that it's possible to import from a 'local' folder
- Display a warning when a 'local' folder is imported, so we warn users on the actual behaviour

screamerbg, 2017-07-05:
This should be now closed, right? @MarceloSalazar

closed 2017-07-05 by screamerbg