ARMmbed/mbed-cli #465: Cannot checkout and deploy branch with lib files that have different remote from master

sarahmarshy, 2017-03-28:
To recreate: 
`mbed import https://github.com/sarahmarshy/mbed-os-example-client-1`

This repo contains a lib file, `easy-connect.lib` that contains `https://github.com/ARMmbed/easy-connect/#6fb5842becaefd360f57916a587b5a0ffdc45ff0`

I have a branch called `c027` where I changed `easy-connect.lib` to reference my fork of easy-connect `https://github.com/sarahmarshy/easy-connect/#87af322ef724cdec08c0f39048020315e7096e67`

Now, if I run:
`git checkout c027` 
`mbed deploy -vv` 

I get this error:
```
C:\Repos\mbed-os-example-client-1>mbed deploy -vv
[mbed] Working path "C:\Repos\mbed-os-example-client-1" (program)
[mbed] Query "git remote -v" in C:\Repos\mbed-os-example-client-1
origin  https://github.com/sarahmarshy/mbed-os-example-client-1 (fetch)
origin  https://github.com/sarahmarshy/mbed-os-example-client-1 (push)
[mbed] Query "git rev-parse HEAD" in C:\Repos\mbed-os-example-client-1
6381035d745f67bfc8dab7a281094b2c0df42504
[mbed] Query "git remote -v" in C:\Repos\mbed-os-example-client-1\easy-connect
origin  https://github.com/ARMmbed/easy-connect (fetch)
origin  https://github.com/ARMmbed/easy-connect (push)
[mbed] Query "git rev-parse HEAD" in C:\Repos\mbed-os-example-client-1\easy-connect
395e3cd97d052c82c51bd63616ea0c4085f45b8b
[mbed] Query "git remote -v" in C:\Repos\mbed-os-example-client-1\easy-connect
origin  https://github.com/ARMmbed/easy-connect (fetch)
origin  https://github.com/ARMmbed/easy-connect (push)
[mbed] Query "git rev-parse HEAD" in C:\Repos\mbed-os-example-client-1\easy-connect
395e3cd97d052c82c51bd63616ea0c4085f45b8b
[mbed] Updating library "easy-connect" to rev #87af322ef724
[mbed] Fetching revisions from remote repository to "easy-connect"
[mbed] Exec "git fetch --all --tags -v" in C:\Repos\mbed-os-example-client-1\easy-connect
Fetching origin
From https://github.com/ARMmbed/easy-connect
 = [up to date]      esp8266-latest -> origin/esp8266-latest
 = [up to date]      master     -> origin/master
 = [up to date]      mbed-client -> origin/mbed-client
 = [up to date]      mbedos5.3.4 -> origin/mbedos5.3.4
 = [up to date]      odin       -> origin/odin
 = [up to date]      revert-mts -> origin/revert-mts
[mbed] Checkout "87af322ef724cdec08c0f39048020315e7096e67" in easy-connect
[mbed] Query "git show-ref" in C:\Repos\mbed-os-example-client-1\easy-connect
5b9cb8cea4a11b0ab974c991b527c9b79fceae75 refs/heads/master
5b9cb8cea4a11b0ab974c991b527c9b79fceae75 refs/remotes/origin/HEAD
0e924fc9504c5fc4425adf30f7ad85a9fe0bf4fd refs/remotes/origin/esp8266-latest
5b9cb8cea4a11b0ab974c991b527c9b79fceae75 refs/remotes/origin/master
723542653307ea78efb2d8b91fc1f69e6d567574 refs/remotes/origin/mbed-client
e61ba698f31f71693451e91f68bb2035706695b7 refs/remotes/origin/mbedos5.3.4
0c77697463c204aa53467a8b7c6286c31b973792 refs/remotes/origin/odin
d3503403331d10000e1d061fe189c17f215d4e89 refs/remotes/origin/revert-mts
[mbed] Exec "git checkout 87af322ef724cdec08c0f39048020315e7096e67" in C:\Repos\mbed-os-example-client-1\easy-connect
fatal: reference is not a tree: 87af322ef724cdec08c0f39048020315e7096e67
[mbed] ERROR: Unable to update "easy-connect" to rev #87af322ef724
---

```

Running with `-vv` it looks like mbed-cli uses the remote in master to fetch the commit, I.E https://github.com/ARMMbed/easy-connect and NOT the URL in the lib file. 

I can work around this by updating the master branch in mbed-os-example-client-1 to reference my personal fork easy-connect, and then I can checkout `c027` branch and deploy without errors.

screamerbg, 2017-03-28:
@sarahmarshy When you want to switch a branch, tag or a hash of already deployed repository (with dependencies), please use `mbed update [branch|tag|hash]`, e.g. `mbed update c027`.

You will get
```
[mbed] Synchronizing dependency references...
[mbed] Updating reference "easy-connect" -> "https://github.com/ARMmbed/mcr20a-rf-driver/#6fb5842becaefd360f57916a587b5a0ffdc45ff0"
[mbed] Updating program "mbed-os-example-client-1" to branch c027
[mbed] Removing library "easy-connect" (changed URL). Will add from new URL.
[mbed] Adding library "easy-connect" from "https://github.com/sarahmarshy/easy-connect" at rev #87af322ef724
[mbed] Adding library "easy-connect/atmel-rf-driver" from "https://github.com/ARMmbed/atmel-rf-driver" at rev #57f22763f4d3
[mbed] Adding library "easy-connect/C027Interface" from "https://mbed.org/teams/NetworkSocketAPI/code/C027Interface" at rev #69ea34ad7fc6
[mbed] Adding library "easy-connect/C027Interface/C027_Support" from "https://developer.mbed.org/teams/NetworkSocketAPI/code/C027_Support" at rev #38b7588c624d
[mbed] Adding library "easy-connect/esp8266-driver" from "https://github.com/ARMmbed/esp8266-driver" at rev #4ed87bf7fe37
[mbed] Adding library "easy-connect/esp8266-driver/ESP8266/ATParser" from "https://github.com/ARMmbed/ATParser" at rev #269f14532b98
[mbed] Adding library "easy-connect/mcr20a-rf-driver" from "https://github.com/ARMmbed/mcr20a-rf-driver" at rev #d8810e105d7d
[mbed] Adding library "easy-connect/stm-spirit1-rf-driver" from "https://github.com/ARMmbed/stm-spirit1-rf-driver" at rev #bbde15ee85d0
[mbed] Updating library "mbed-client" to rev #52e65b46dff7
[mbed] Updating library "mbed-client/mbed-client-c" to rev #c739b8cbcc57
[mbed] Updating library "mbed-client/mbed-client-classic" to rev #b9a521dcd0fc
[mbed] Updating library "mbed-client/mbed-client-mbed-tls" to rev #7e1b6d815038
[mbed] Updating library "mbed-os" to rev #f4864dc6429e
[mbed] Updating library "pal" to rev #4e46c0ea8706
```

Notice how mbed CLI removed a repository and replaced it with a different one that you specified.

Also note that `mbed deploy` is only used for initial importing. Generally `mbed import` and `mbed update` is the preferred workflow as mbed CLI knows how to handle changes of repositories (e.g. replace one repository with another but keep the same URL).

Hope this makes sense.

screamerbg, 2017-03-28:
Also see the `mbed update` guide at https://github.com/ARMmbed/mbed-cli#updating-to-an-upstream-version

sarahmarshy, 2017-03-28:
Oh, awesome! Thanks. I'll close this now.

closed 2017-03-28 by sarahmarshy