ARMmbed/mbed-cli #467: Reduce visual clutter in top level directory

wdwalker, 2017-03-30:
### Description
- Type: Enhancement
- Priority: Minor

## Enhancement

**Reason to enhance or problem with existing solution**

When I use the mbed CLI to work with mbed OS 5.4, I end up with a number of files in the top level directory that are for mbed itself: mbed-os.lib, mbed-os, mbed_settings.py[c], a directory and *.lib pair for each library I import, mbed_app.json for my app, etc. In addition, there are other hidden files, such as .mbed and .mbedignore.

It would be great to reduce this clutter.

**Suggested enhancement**

Make a hidden subdirectory to hold the mbed configs. I don't have this well thought out -- were it not for the existence of the .mbed file, I'd suggest making a .mbed directory to hold all of the mbed configs, which might be considered the mbed equivalent of the .git directory.

I've marked this as "minor" because the impact is mostly related to a minor facet of developer experience and has no impact on the functionality of an application. I will not be offended at all if this enhancement request is rejected or closed as "won't fix".

**Pros**

Visual clutter in the top-level directory is reduced.

**Cons**

Hiding things can lead to confusion. This may also require a large change to the mbed CLI just for the benefit of reducing clutter.

screamerbg, 2017-03-31:
@wdwalker thanks for the suggestions.

Here are a few more cons:
* .lib in "special" folder means that all the libraries must be in the root folder of the parent repository. Currently .lib can live in subfolders and therefore libraries are imported where these files are. Also this will hide whether a subfolder is actually a subrepository (currently is easy to tell by checking whether there's .lib file alongside the folder)
* .mbedignore is relative to the folder it's placed. Having this file in a different folder means that the mbed OS build system need to handle a different location than the folder it's intended to. This is also inconsistent with .gitignore/.hgignore behavior.

Same as you I'm not convinced whether the reduction of clutter justifies hiding things + possible confusion + changing logic + general inconsistency.

wdwalker, 2017-03-31:
I'm happy to withdraw this request. My main motivation is that I have to visually scan through all the clutter to find the things I really care about (e.g., README.md, my sources, etc.).  Right now, working with mbed CLI feels a bit like working directly in a .git directory. If .git didn't exist, imaging see all of this each time you did an 'ls' at the top level directory:

HEAD		description	info		packed-refs
branches	hooks		logs		refs
config		index		src		objects

Buried in all that is my 'src' directory, which I've created as a best practice of using consistent directory names to hold the stuff I'm working on vs. the stuff that helps me work.

theotherjimmy, 2017-11-28:
@wdwalker I don't think that we are going to change this. I'm closing this issue as "Won't do"

closed 2017-11-28 by theotherjimmy