ARMmbed/mbed-cli #468: Cannot export LPC1768 to gnuarmeclipse

wdwalker, 2017-03-30:
## Bug

**Target**
LPC1768

**Toolchain:**
GCC_ARM

**Toolchain version:**
(GNU Tools for ARM Embedded Processors 6-2017-q1-update) 6.3.1 20170215 (release) [ARM/embedded-6-branch revision 245512]

**mbed-cli version:** 
1.0.0

**mbed-os sha:**
067fe9b Merge pull request #3848 from jamike/USBAUDIO_JITTER

**Expected behavior**
The following command should result in a GNU ARM Eclipse environment for the LPC1768:
<code>
mbed export -m LPC1768 -I gnuarmeclipse
</code>

**Actual behavior**
I get the following error:
<code>
usage: project.py [-h] [-m MCU] [-i IDE] [-c] [-p PROGRAM] [-n PROGRAM] [-b]
                  [-L] [-S] [-E] [--source SOURCE_DIR] [-D MACROS]
                  [--profile PROFILE] [--update-packs]
project.py: error: LPC1768 not supported by gnuarmeclipse
[mbed] ERROR: "python" returned error code 2.
[mbed] ERROR: Command "python -u /Users/wwalker/work/github/wdwalker/HelloMbedStepper/mbed-os/tools/project.py -i gnuarmeclipse -m LPC1768 --source ." in "/Users/wwalker/work/github/wdwalker/HelloMbedStepper"
</code>

**Steps to reproduce**
Run the command listed in "Expected behavior"

wdwalker, 2017-03-30:
The root cause of the seems to be the POST_BINARY_WHITELIST in tools/export/gnuarmeclipse/__init__.py not containing "LPCTargetCode.lpc_patch".

"LPCTargetCode.lpc_patch" is referred to in the "LPCTarget" section of targets.json, and its intended function is to address 32.3.1.1 Criterion for Valid User Code from the NXP LPC1768 User Manual.

Without this patch, OpenOCD complains with a warning when gdb attempts to load the binary:
<code>
"Warn : Verification will fail since checksum in image (0x00000000) to be written to flash is different from calculated vector checksum (0xefff70e6)."
"Warn : To remove this warning modify build tools on developer PC to inject correct LPC vector checksum."
</code>

Note that things seem to work fine even in the face of this warning.

wdwalker, 2017-03-30:
There are several ways to address this problem. I believe the ultimate solution is to get the exporters to actually do something with the post_binary_hooks. Right now, it doesn't look like the exporters do anything with them except use them as a means for rejection.

The immediate problem can be resolved by adding "LPCTargetCode.lpc_patch" to the POST_BINARY_WHITELIST in tools/export/gnuarmeclipse/\_\_init\_\_.py, and I believe this should be fine. Might also want to do the same for the embitz and makefile exporters.

Following the above, the exporters should be modified to actually do something with the post_binary_hook (e.g., gnuarmeclipse could add a postannouncebuildStep to .cproject), but that's a separate issue.

theotherjimmy, 2017-03-30:
Hey @wdwalker. Thanks for providing exceptional detail on the issue at hand. It seems like you know exactly what to do.

> The immediate problem can be resolved by adding "LPCTargetCode.lpc_patch" to the POST_BINARY_WHITELIST 

I would approve this patch.

> Following the above, the exporters should be modified to actually do something with the post_binary_hook (e.g., gnuarmeclipse could add a postannouncebuildStep to .cproject), but that's a separate issue.

I recommend that we do the aforementioned patch first, and see if we can make post binary hooks more "exportable".

wdwalker, 2017-03-30:
Thank you. I have a patch ready to go and I've tested it merely by exporting to IDEs that will exercise the changed code path (see below). I'm struggling with testing per [https://github.com/ARMmbed/mbed-cli#testing](https://github.com/ARMmbed/mbed-cli#testing) - I've never used the test feature before and I'm likely struggling from pilot error.

I'm also not sure how to actually do a pull request with mbed-cli. I'm used to gitflow workflow and working off of feature branches, but I don't see a way to do that here. Any advice would be great - for example, should I just attached a 'git diff' patch to this issue?

Williams-MacBook-Pro:HelloMbedStepper wwalker$ mbed export -m LPC1768 -i gnuarmeclipse
Scan: .
Scan: FEATURE_BLE
Scan: FEATURE_LWIP
Scan: FEATURE_ETHERNET_HOST
Scan: FEATURE_LOWPAN_BORDER_ROUTER
Scan: FEATURE_LOWPAN_HOST
Scan: FEATURE_LOWPAN_ROUTER
Scan: FEATURE_NANOSTACK
Scan: FEATURE_NANOSTACK_FULL
Scan: FEATURE_THREAD_BORDER_ROUTER
Scan: FEATURE_THREAD_END_DEVICE
Scan: FEATURE_THREAD_ROUTER

Create a GNU ARM Eclipse C++ managed project
Project name: HelloMbedStepper
Target: LPC1768
Toolchain: GCC_ARM
Source folders: 13, with 56 exclusions
Include folders: 36
Symbols: 38
Linker script: mbed-os/targets/TARGET_NXP/TARGET_LPC176X/device/TOOLCHAIN_GCC_ARM/LPC1768.ld

Build configuration: Debug
Common flags: -c -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -O0 -g3 -mcpu=cortex-m3 -mthumb
C++ flags: -std=gnu++98 -fno-rtti -Wvla -include mbed_config.h
C flags: -std=gnu99 -include mbed_config.h
ASM flags: -x assembler-with-cpp
Linker flags: -Wl,--gc-sections -Wl,--wrap,main -Wl,--wrap,_malloc_r -Wl,--wrap,_free_r -Wl,--wrap,_realloc_r -Wl,--wrap,_calloc_r -Wl,--wrap,exit -Wl,--wrap,atexit -Wl,-n -mcpu=cortex-m3 -mthumb

Build configuration: Develop
Common flags: -c -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -Os -mcpu=cortex-m3 -mthumb
C++ flags: -std=gnu++98 -fno-rtti -Wvla -include mbed_config.h
C flags: -std=gnu99 -include mbed_config.h
ASM flags: -x assembler-with-cpp
Linker flags: -Wl,--gc-sections -Wl,--wrap,main -Wl,--wrap,_malloc_r -Wl,--wrap,_free_r -Wl,--wrap,_realloc_r -Wl,--wrap,_calloc_r -Wl,--wrap,exit -Wl,--wrap,atexit -Wl,-n -mcpu=cortex-m3 -mthumb

Build configuration: Release
Common flags: -c -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -Os -DNDEBUG -mcpu=cortex-m3 -mthumb
C++ flags: -std=gnu++98 -fno-rtti -Wvla -include mbed_config.h
C flags: -std=gnu99 -include mbed_config.h
ASM flags: -x assembler-with-cpp
Linker flags: -Wl,--gc-sections -Wl,--wrap,main -Wl,--wrap,_malloc_r -Wl,--wrap,_free_r -Wl,--wrap,_realloc_r -Wl,--wrap,_calloc_r -Wl,--wrap,exit -Wl,--wrap,atexit -Wl,-n -mcpu=cortex-m3 -mthumb

Done. Import the 'HelloMbedStepper' project in Eclipse.
Williams-MacBook-Pro:HelloMbedStepper wwalker$ mbed export -m LPC1768 -i eclipse_gcc_arm
Scan: .
Scan: FEATURE_BLE
Scan: FEATURE_LWIP
Scan: FEATURE_ETHERNET_HOST
Scan: FEATURE_LOWPAN_BORDER_ROUTER
Scan: FEATURE_LOWPAN_HOST
Scan: FEATURE_LOWPAN_ROUTER
Scan: FEATURE_NANOSTACK
Scan: FEATURE_NANOSTACK_FULL
Scan: FEATURE_THREAD_BORDER_ROUTER
Scan: FEATURE_THREAD_END_DEVICE
Scan: FEATURE_THREAD_ROUTER
Williams-MacBook-Pro:HelloMbedStepper wwalker$ mbed export -m LPC1768 -i make_gcc_arm
Scan: .
Scan: FEATURE_BLE
Scan: FEATURE_LWIP
Scan: FEATURE_ETHERNET_HOST
Scan: FEATURE_LOWPAN_BORDER_ROUTER
Scan: FEATURE_LOWPAN_HOST
Scan: FEATURE_LOWPAN_ROUTER
Scan: FEATURE_NANOSTACK
Scan: FEATURE_NANOSTACK_FULL
Scan: FEATURE_THREAD_BORDER_ROUTER
Scan: FEATURE_THREAD_END_DEVICE
Scan: FEATURE_THREAD_ROUTER
Williams-MacBook-Pro:HelloMbedStepper wwalker$ mbed export -m LPC1768 -i gcc_arm
Scan: .
Scan: FEATURE_BLE
Scan: FEATURE_LWIP
Scan: FEATURE_ETHERNET_HOST
Scan: FEATURE_LOWPAN_BORDER_ROUTER
Scan: FEATURE_LOWPAN_HOST
Scan: FEATURE_LOWPAN_ROUTER
Scan: FEATURE_NANOSTACK
Scan: FEATURE_NANOSTACK_FULL
Scan: FEATURE_THREAD_BORDER_ROUTER
Scan: FEATURE_THREAD_END_DEVICE
Scan: FEATURE_THREAD_ROUTER
Williams-MacBook-Pro:HelloMbedStepper wwalker$ mbed export -m LPC1768 -i embitz
Scan: .
Scan: FEATURE_BLE
Scan: FEATURE_LWIP
Scan: FEATURE_ETHERNET_HOST
Scan: FEATURE_LOWPAN_BORDER_ROUTER
Scan: FEATURE_LOWPAN_HOST
Scan: FEATURE_LOWPAN_ROUTER
Scan: FEATURE_NANOSTACK
Scan: FEATURE_NANOSTACK_FULL
Scan: FEATURE_THREAD_BORDER_ROUTER
Scan: FEATURE_THREAD_END_DEVICE
Scan: FEATURE_THREAD_ROUTER
Williams-MacBook-Pro:HelloMbedStepper wwalker$

wdwalker, 2017-03-30:
Here's a patch file created with "git diff" just in case:

[patch.txt](https://github.com/ARMmbed/mbed-cli/files/883515/patch.txt)

theotherjimmy, 2017-03-30:
>I'm struggling with testing per https://github.com/ARMmbed/mbed-cli#testing - I've never used the test feature before and I'm likely struggling from pilot error.

We test exporters a bit differently. I'll comment in a moment with how that works from a local machine. We run /morph export-bulid on a PR that modifies the exporters.

> I'm used to gitflow workflow and working off of feature branches, but I don't see a way to do that here.

Fork the mbed-os repo (that's where your modifications live), and push your changes into a branch on that fork. After that, you should be able to submit a PR using the "Compare and Pull Request" Banner on the mbed-os repo home page.

theotherjimmy, 2017-03-30:
You beet me to it. :laughing: 

I can submit the changes as a PR if you would like, but I would prefer that you have credit and the ability to modify the change set in the future if issues come up.

wdwalker, 2017-03-30:
Aha - OK. I'll set up a fork and set up the PR. Many thanks for the help.

theotherjimmy, 2017-03-30:
Thank you for contributing!

wdwalker, 2017-03-30:
BTW - I've opened issue #469 for the "exporter friendly" topic.

closed 2017-04-06 by sg-