ARMmbed/mbed-cli #469: Post binary hooks are not called by exporters

wdwalker, 2017-03-30:
## Bug

**Target**
LPC1768

**Toolchain:**
GCC_ARM

**Toolchain version:**
(GNU Tools for ARM Embedded Processors 6-2017-q1-update) 6.3.1 20170215 (release) [ARM/embedded-6-branch revision 245512]

**mbed-cli version:** 
1.0.0

**mbed-os sha:**
067fe9b Merge pull request #3848 from jamike/USBAUDIO_JITTER

**Expected behavior**
This is related to issue #468. Right now, it seems as though exporters do not call post binary hooks. For example, with the fix for issue #468, we get a GNU ARM Eclipse environment for the LPC1768 using the following command:
<code>
mbed export -m LPC1768 -i gnuarmeclipse
</code>

I would expect the resulting GNU ARM Eclipse environment to run the post binary hook (LPCTargetCode.lpc_patch) when it creates the *.bin file. 

**Actual behavior**
The resulting GNU ARM Eclipse environment does not run the post binary hook (LPCTargetCode.lpc_patch) when it creates the *.bin file. 

**Steps to reproduce**
Run the command listed in "Expected behavior"

wdwalker, 2017-03-30:
While each exporter is likely going to need to do its own thing, perhaps the mbed-cli could be made exporter friendly - perhaps by adding a "apply-post-binary-hook" option or something that would take the binary file and run the post_binary_hook on it.

theotherjimmy, 2017-03-30:
When you export, there is no guarantee that you still have the mbed-os tools directory at all. This makes adding an "apply-post-binary-hook" option problematic. Further, the post binary hooks are currently set up as `class.method` names that have the option to import arbitrary mbed-os tools code.

theotherjimmy, 2017-03-30:
This issue is really related to the mbed-os tools directory, but I have no problem tracking it here.

wdwalker, 2017-03-30:
I'm unsure of the expected workflow when using an exporter. My mental model is that I'm going to use GNU ARM Eclipse for development - edit, compile, debug, etc. - but I'm going to continue to have a cloned mbed-os repo in a subdir as well as subdirs for all other libraries I've added. So, I should always have a mbed-os tools directory.

The other mental model I have right now is that I may end up running an exporter several times in my app repo so I can work on the same code, but with different boards (NXP, ST Micro, etc.). This is a little clunky at the moment with GNU ARM Eclipse.

theotherjimmy, 2017-03-30:
The exporters are also presented as a service on developer.mbed.org.

wdwalker, 2017-03-30:
Aha - OK. I understand now. Thanks.

theotherjimmy, 2017-11-28:
Please reopen if this is still a problem

closed 2017-11-28 by theotherjimmy