ARMmbed/mbed-cli #471: mbed-cli ignores config when mbed_app.json invalid

bremoran, 2017-03-30:
mbed-cli builds anyway when mbed_app.json is invalid, which leads to unexpected debugging errors.

screamerbg, 2017-03-30:
cc @theotherjimmy

teetak01, 2017-03-31:
I also noticed the exactly same thing yesterday. I copied some defines from git and had something like this in mbed_app.json:

```
   "macros": ["MACRO1",
               "MACRO2",
+               "MACRO3"],
```

So I had a `+` sign there, which I did not notice for a while and got _very_ strange fails in compilation :-)

Maybe mbed-cli could do some validation for critical configuration files?

theotherjimmy, 2017-03-31:
Okay guys, mbed tools do report the error, it's just hidden by tonnes of output.
For example:
1. `cat mbed_app.json` : `,`
2. `mbed compile`
```
Error parsing './mbed_app.json':
No JSON object could be decoded
Building project mbed-os-example-fat-filesystem (K64F, GCC_ARM)
Scan: .
Scan: FEATURE_LWIP
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_UVISOR
Scan: FEATURE_BLE
Scan: FEATURE_LOWPAN_ROUTER
Scan: FEATURE_ETHERNET_HOST
Scan: FEATURE_NANOSTACK
Scan: FEATURE_THREAD_END_DEVICE
Scan: FEATURE_THREAD_ROUTER
Scan: FEATURE_THREAD_BORDER_ROUTER
Scan: FEATURE_LOWPAN_BORDER_ROUTER
Scan: FEATURE_NANOSTACK_FULL
Scan: FEATURE_LOWPAN_HOST
Scan: FEATURE_STORAGE
Scan: mbed
Scan: env
```

theotherjimmy, 2017-03-31:
I take it that you are requesting that it stops and does not continue.

theotherjimmy, 2017-03-31:
See ARMmbed/mbed-os#4098 for a fix.

closed 2017-04-07 by adbridge