ARMmbed/mbed-cli #479: mbed-cli leaks compiler process in CI

mazimkhan, 2017-04-21:
Please see the screenshots:
[image]
[image]

Suspended/Zombi ```arm-none-eabi-g++.exe``` prevent recursive deletion of workspace. Causing subsequent builds to fail.

theotherjimmy, 2017-04-21:
This is a problem with the tools in mbed-os. Please re-open there.

theotherjimmy, 2017-04-21:
Further, it should not be a problem if you have an init process that reaps orphaned children.

mazimkhan, 2017-04-21:
I would not want to pollute CI jobs with cleaning up of orphaned processes. Unless there is a slick Jenkins plugin that does this reliably.
It would be better at least maintained apps like build tools don't leak processes.

mazimkhan, 2017-04-21:
Created https://github.com/ARMmbed/mbed-os/issues/4213

jupe, 2017-04-22:
CC: @miklis @marhil01

screamerbg, 2017-06-21:
This is now tracked in https://github.com/ARMmbed/mbed-os/issues/4213. Closing here

closed 2017-06-21 by screamerbg