ARMmbed/mbed-cli #480: Make mbed-cli more script friendly

seppestas, 2017-04-25:
Mbed allows you to store and get some configuration values using `mbed config` and special purpose commands like `mbed target` and `mbed toolchain`. However, the output of these commands when getting the configured value is for some reason always prepended with `[mbed] `, making it a lot harder to use these values in a scripting environment like e.g a Makefile.

Example: a script to flash the MCU needs to know the binary that is the output of the mbed compiler. This binary is typically located in `BUILD/<mbed target>/<mbed toolchain>/`. However there is no way for the flash script to simply find the binary by using something like ``BUILD/`mbed target`/`mbed toolchain`/$ARTIFACT_NAME`` because `mbed target` outputs `[mbed] MBED_TARGET` instead of `MBED_TARGET`. The same goes for `mbed toolchain` and `mbed config [TOOLCHAIN|TARGET|...]`.

Having these command just output the value would make them a lot more useful for scripting use cases.

bridadan, 2017-04-25:
@seppestas Fair point, most CLI tools usually have a "machine readable" option that removes the extra human-friendly formatting. I don't think this would be too hard to implement, but it's not as big of a priority. Feel free to give it a shot 😄 

If you're already using scripting, you could always ignore the `[mbed]` portion with a regular expression or something, but I can see how this would make the script more complex.

seppestas, 2017-04-25:
Could the `[mbed]` appendix just be ditched? IMO it doesn't add a lot in terms of human-friendliness.

bridadan, 2017-04-25:
I believe the motivation for having the `[mbed]` in the cli is because mbed CLI actually calls multiple different scripts and tools (the python scripts inside mbed OS, greentea for tests, mbedls, etc). When examining logs, it's pretty crucial to know what script is printing what information. For this reason I'd prefer not to remove it. But I'm not the only stakeholder here, so other people can weigh in as well.

seppestas, 2017-04-25:
I think this feature is mostly used when debugging the tools, in which case it could be made available when providing the verbose flag or some kind of debug flag.

bridadan, 2017-04-25:
Seems like a fair suggestion to me!

@screamerbg any thoughts?

screamerbg, 2017-05-03:
@seppestas as @bridadan put it, mbed CLI invokes a lot of tools - git, hg/mercurial, build system, exporters, test frameworks, detect scripts etc. We would like to make it very clear who "owns" the error, rather than mask it behind mbed CLI.

One of the future updates to mbed CLI will introduce a prefix to each line with the tool name that emits the output. This is align the output of mbed CLI and mbed Greentea.

To your question, we're not planning to deprecate the prefix, but rather make it even more obvious who generates the output. With that addition, we would also add the ability to remove the prefix for when you want to process it with other scripts and need a stable output, e.g. see git's --porcelain flag

closed 2019-10-31 by ciarmcom