ARMmbed/mbed-cli #481: mbed compile does not fail when failing to parse mbed_app.json

seppestas, 2017-04-25:
When there is invalid json in the mbed_app.json file, `mbed compile` will display an error message like e.g
```
Error parsing './mbed_app.json':
Expecting property name: line 14 column 1 (char 262)
```
However, it will not cause the command to fail, resulting in a binary that does not contain any of the configuration set in mbed_app.json and the error message to get lost in the output of the compilation. This is a problem that can be easily triggered by e.g improperly removing an entry in the `target_overrides` resulting in a trailing comma and thus invalid json.

Not failing on parsing the mbed_app.json file is also inconsistent with other errors related to the configuration like e.g overriding an undefined parameter, which causes `mbed compile` to fail with an error message like
```
[ERROR] Attempt to override undefined parameter 'target.fubar' in 'application[SOME_TARGET]'
```

bridadan, 2017-04-25:
@seppestas Thanks for raising this issue!

@theotherjimmy Any thoughts on this? It seems like we'd probably want to fail ASAP.

theotherjimmy, 2017-04-25:
https://github.com/ARMmbed/mbed-os/pull/4098 should have fixed it. What version of mbed OS are you using?

bridadan, 2017-04-25:
According to that PR, it should have made it into the 5.4.3 release, which is the latest release at the time of writing.

theotherjimmy, 2017-04-25:
That's why I was asking which version of mbed OS. If you are using an older version or an mbed OS 2 version, you may not have that patch yet. For mbed OS 2 I can make that happen.

seppestas, 2017-04-25:
I'm currently using mbed os 5.4.2 and the nrf5_dfu_s13x_v2 feature branch (33bf0a8b5384f205fe315466bf798b4c85aa11f1).

I'll see if it works on mbed OS 5.4.3.

theotherjimmy, 2017-04-25:
You could also rebase the `nrf5_dfu_s13x_v2` feature branch onto the current master (I'm supposing that it was based on master at one point). That might be difficult, so be warned.

screamerbg, 2017-05-03:
@seppestas Any news on this? Did @theotherjimmy's suggestion help?

seppestas, 2017-05-03:
@screamerbg Oops, sorry, forgot about this...

Confirmed to be solved in mbed-os 5.4.4 by ARMmbed/mbed-os#4098.
This is a duplicate of ARMmbed/mbed-cli#471.

closed 2017-05-03 by seppestas