ARMmbed/mbed-cli #482: No way to turn off mbed cache!

JanneKiiskila, 2017-04-25:
The documentation at least is unclear OR the possibility to turn off the mbed cache is missing.
```
jankii01@ubuntu:~/mbed/mbed-cloud-client-example$ mbed config cache none
[mbed] none now set as default cache in program "mbed-cloud-client-example"
```
So, how do you turn it off? 

One needs to turn it off, because currently the cache is failing randomly and you get a completely random SW  package which either compiles or not or fails to work. 👎

bridadan, 2017-04-25:
According to the readme, you're supposed to be able to set the option to `none` and that should disable it: https://github.com/ARMmbed/mbed-cli#mbed-cli-configuration

Does it look like CLI isn't actually disabling the cache at this point? Can you give us the steps to reproduce the issue?

JanneKiiskila, 2017-04-25:
No, I think the mbed config --global cache none does the trick, but the UI isn't perhaps the most explicit one. I looked into the code as well and it seem 3 different values would do the trick; 

https://github.com/ARMmbed/mbed-cli/blob/master/mbed/mbed.py#L886-L889

none, off and disabled. Perhaps we could just modify the README.MD a bit to clarify how to turn it off. Now it's a bit "assumed" implicitly there.

JanneKiiskila, 2017-04-25:
Apparently the cache issue was also fixed, but it just hasn't been released? We should really release the fix, if we've fixed the issue.

screamerbg, 2017-04-25:
@JanneKiiskila the cache issues were fixed. Did you update the latest mbed CLI? If not, care to update more often?

JanneKiiskila, 2017-04-25:
Indeed, update done - thanks for the tip, I was clearly behind - my apologies for that (0.9.1). Is there a way to subscribe for release notes? 

Anyways, PR done to add the addition to README.md; https://github.com/ARMmbed/mbed-cli/pull/483

bridadan, 2017-04-25:
@JanneKiiskila I don't think there's anything like that as of this moment. You could always watch the mbed CLI repo (though you'll get other emails as well).

@screamerbg Release notes would probably be useful to send across the team, similar to what we do for mbed OS releases.

JanneKiiskila, 2017-04-25:
PR merged, issue closed.

closed 2017-04-25 by JanneKiiskila