ARMmbed/mbed-cli #484: Having a BUILD folder in further down in your path produces "[ERROR] header.tmpl"

JanneKiiskila, 2017-04-26:
This me took me hours to figure, kudos to @terojaasko for pointing it out to me. 
If you accidentally do `mbed compile` in a folder down your dependency repos you will get a weird  error.

{code}
jankii01@ubuntu:~/mbed/mbed-cloud-client-example$ mbed compile -m K64F -t GCC_ARM
[mbed] WARNING: Could not find mbed program in current path "/home/jankii01/mbed/mbed-cloud-client-example".
[mbed] WARNING: You can fix this by calling "mbed new ." in the root of your program.
---
Building project mbed-cloud-client-example (K64F, GCC_ARM)
Scan: .
Scan: FEATURE_LWIP
Scan: FEATURE_UVISOR
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_BLE
Scan: FEATURE_STORAGE
Scan: FEATURE_THREAD_BORDER_ROUTER
Scan: FEATURE_THREAD_ROUTER
Scan: FEATURE_LOWPAN_BORDER_ROUTER
Scan: FEATURE_LOWPAN_ROUTER
Scan: FEATURE_LOWPAN_HOST
Scan: FEATURE_NANOSTACK_FULL
Scan: FEATURE_NANOSTACK
Scan: FEATURE_ETHERNET_HOST
Scan: FEATURE_THREAD_END_DEVICE
Scan: env
Scan: mbed
[ERROR] header.tmpl
[mbed] ERROR: "/usr/bin/python" returned error code 1.
[mbed] ERROR: Command "/usr/bin/python -u /home/jankii01/mbed/mbed-cloud-client-example/mbed-os/tools/make.py -t GCC_ARM -m K64F --source . --build ./BUILD/K64F/GCC_ARM" in "/home/jankii01/mbed/mbed-cloud-client-example"
---
jankii01@ubuntu:~/mbed/mbed-cloud-client-example$ find . -name "BUILD"
./mbed-os/BUILD
./mbed-cloud-client/factory_client/mbed-client-esfs/Test/BUILD
./BUILD
{code}

This is caused by having a BUILD folder anywhere down the downstream folders. It just kills the compilation and the error note is NOT very clear in giving any sensible guidance how to recover.

0xc0170, 2017-04-26:
I had this issue yesterday, could not figure out where is that header coming from. But as I cleaned the repo afterwards, I could not reproduce this further to report  it back.

JanneKiiskila, 2017-04-26:
Yep, it required then `mbed update --clean` to get fully resolved.

bridadan, 2017-04-26:
Hmm, this seems funny. I thought mbed CLI was supposed to recursively traverse upwards in your project to try and find the root. This behavior usually happens when you've imported the project in a funny way so that the "root" for the project was never setup correctly by mbed CLI. If you have steps that detail how to setup your project exactly like you have here from scratch, I'd be interested in trying to reproduce this. Thanks!

0xc0170, 2017-04-28:
My use case how I got this error:

Having blinky example for mbed OS, and inside mbed-os repo . 
I am working on both codebases. Using mbed-cli to build blinky with mbed OS or any other program with main.cpp and then inside mbed-os repository, running tools/mbed 2 scripts like make.py or build.py to verify that changes I am doing do not affect mbed 2 scripts + programs. This is however not typical use case, at least  the error could be different, I just cleaned all my repo (wiped any stashed or unstashed files to get it working and continued).

theotherjimmy, 2017-04-28:
When I follow the instructions you posted, @0xc0170, I get the following error:
```[ERROR] Library name 'platform' is not unique (defined in '/home/jimbri01/src/armmbed/mbed-os-example-fat-filesystem/mbed-os/platform/mbed_lib.json' and '/home/jimbri01/src/armmbed/mbed-os-example-fat-filesystem/mbed-os/BUILD/mbed/TARGET_K64F/mbed_lib.json')
```

0xc0170, 2017-04-28:
That's correct, now I get the same error as you:

```
[ERROR] Library name 'platform' is not unique (defined in '\mbed-os-example-blinky\mbed-os\platform\mbed_lib.json' and 'C:\Code\examples\mbed-os-example-blinky\mbed-os\BUILD\mbed\TARGET_K64F\mbed_lib.json')
```

in the blinky root and the same one for the ``mbed test -m K64F -t ARM --list`` in blinky/mbed-os.  😕 

(Mine above instructions were what I was doing most of the time and got the same error as reported above multiple times, seems like I did something additional, will continue and if I hit it again, will report back).

screamerbg, 2017-05-03:
@0xc0170 @JanneKiiskila could you please update your mbed CLI to latest (v1.1.1) and provide steps to reproduce it with it?

ashok-rao, 2017-08-02:
@screamerbg .. I faced this issue as well today with 1.1.1 . Steps below:
1. mbed new .
2. created a main.cpp with some lines.
3. cd mbed-os
4. checkout a particular SHA.
5. mbed compile -t ARM -m ublox_evk_nina_b1

I got this:
```
C:\Ashok\Sensize>mbed compile -t ARM -m ublox_evk_nina_b1 -c
Building project Sensize (UBLOX_EVK_NINA_B1, ARM)
Scan: .
Scan: FEATURE_BLE
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_LWIP
Scan: FEATURE_UVISOR
Scan: FEATURE_ETHERNET_HOST
Scan: FEATURE_LOWPAN_BORDER_ROUTER
Scan: FEATURE_LOWPAN_HOST
Scan: FEATURE_LOWPAN_ROUTER
Scan: FEATURE_NANOSTACK
Scan: FEATURE_NANOSTACK_FULL
Scan: FEATURE_THREAD_BORDER_ROUTER
Scan: FEATURE_THREAD_END_DEVICE
Scan: FEATURE_THREAD_ROUTER
Scan: FEATURE_STORAGE
Scan: mbed
Scan: env
[ERROR] header.tmpl
[mbed] ERROR: "c:\python27\python.exe" returned error code 1.
[mbed] ERROR: Command "c:\python27\python.exe -u C:\Ashok\Sensize\mbed-os\tools\make.py -t ARM -m ublox_evk_nina_b1 --source . --build .\BUILD\ublox_evk_nina_b1\ARM -c" in "C:\Ashok\Sensize"
---

C:\Ashok\Sensize>mbed compile -t ARM -m ublox_evk_nina_b1
Building project Sensize (UBLOX_EVK_NINA_B1, ARM)
Scan: .
Scan: FEATURE_BLE
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_LWIP
Scan: FEATURE_UVISOR
Scan: FEATURE_ETHERNET_HOST
Scan: FEATURE_LOWPAN_BORDER_ROUTER
Scan: FEATURE_LOWPAN_HOST
Scan: FEATURE_LOWPAN_ROUTER
Scan: FEATURE_NANOSTACK
Scan: FEATURE_NANOSTACK_FULL
Scan: FEATURE_THREAD_BORDER_ROUTER
Scan: FEATURE_THREAD_END_DEVICE
Scan: FEATURE_THREAD_ROUTER
Scan: FEATURE_STORAGE
Scan: mbed
Scan: env
[ERROR] header.tmpl
[mbed] ERROR: "c:\python27\python.exe" returned error code 1.
[mbed] ERROR: Command "c:\python27\python.exe -u C:\Ashok\Sensize\mbed-os\tools\make.py -t ARM -m ublox_evk_nina_b1 --source . --build .\BUILD\ublox_evk_nina_b1\ARM" in "C:\Ashok\Sensize"
---

C:\Ashok\Sensize>mbed --version
1.1.1
```

theotherjimmy, 2017-11-28:
@JanneKiiskila @ashok-rao I cannot reproduce the failure. Please reopen if this is still a problem.

closed 2017-11-28 by theotherjimmy