ARMmbed/mbed-cli #485: Error while installing mbed-cli on macOS 10.12.4

dhamukrish, 2017-04-26:
Could you please advise why this error as I am very new to mbed world installing on the macOS 10.12.4.  

Below command, I have tried based on the instruction given by mbed

**command**
```sudo pip install -r requirements.txt```

```
Mac:mbed-os-example-blinky username$ mbed compile -t GCC_ARM -m NUCLEO_L073RZ
[mbed] Auto-installing missing Python modules...
[mbed] WARNING: Unable to auto-install required Python modules.
---
[mbed] WARNING: -----------------------------------------------------------------
[mbed] WARNING: The mbed OS tools in this program require the following Python modules: mbed_host_tests, mbed_greentea, beautifulsoup4, fuzzywuzzy
[mbed] WARNING: You can install all missing modules by running "pip install -r requirements.txt" in "/Users/username/Documents/mbed-os-example-blinky/mbed-os"
[mbed] WARNING: On Posix systems (Linux, Mac, etc) you might have to switch to superuser account or use "sudo"
---
Traceback (most recent call last):
  File "/Users/username/Documents/mbed-os-example-blinky/mbed-os/tools/make.py", line 45, in <module>
    from tools.options import get_default_options_parser
  File "/Users/username/Documents/mbed-os-example-blinky/mbed-os/tools/options.py", line 21, in <module>
    from tools.toolchains import TOOLCHAINS
  File "/Users/username/Documents/mbed-os-example-blinky/mbed-os/tools/toolchains/__init__.py", line 28, in <module>
    from tools.config import Config
  File "/Users/username/Documents/mbed-os-example-blinky/mbed-os/tools/config.py", line 26, in <module>
    from tools.arm_pack_manager import Cache
  File "/Users/username/Documents/mbed-os-example-blinky/mbed-os/tools/arm_pack_manager/__init__.py", line 2, in <module>
    from bs4 import BeautifulSoup
ImportError: No module named bs4
[mbed] ERROR: "/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python" returned error code 1.
[mbed] ERROR: Command "/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python -u /Users/username/Documents/mbed-os-example-blinky/mbed-os/tools/make.py -t GCC_ARM -m NUCLEO_L073RZ --source . --build ./BUILD/NUCLEO_L073RZ/GCC_ARM" in "/Users/username/Documents/mbed-os-example-blinky"
```

**Error**
```
mbed-host-tests>=1.1.2->-r requirements.txt (line 10))
Requirement already satisfied (use --upgrade to upgrade): pyparsing in /System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python (from packaging>=16.8->setuptools>=19.0->hidapi->pyOCD>=0.8.1a1->mbed-host-tests>=1.1.2->-r requirements.txt (line 10))
Installing collected packages: setuptools, hidapi, pyOCD, mbed-host-tests, lockfile, funcsigs, pbr, mock, mbed-greentea, beautifulsoup4, fuzzywuzzy
  Found existing installation: setuptools 18.5
    Uninstalling setuptools-18.5:
Exception:
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pip/basecommand.py", line 211, in main
    status = self.run(options, args)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pip/commands/install.py", line 311, in run
    root=options.root_path,
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pip/req/req_set.py", line 640, in install
    requirement.uninstall(auto_confirm=True)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pip/req/req_install.py", line 716, in uninstall
    paths_to_remove.remove(auto_confirm)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pip/req/req_uninstall.py", line 125, in remove
    renames(path, new_path)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pip/utils/__init__.py", line 315, in renames
    shutil.move(old, new)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/shutil.py", line 299, in move
    copytree(src, real_dst, symlinks=True)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/shutil.py", line 208, in copytree
    raise Error, errors
Error: [('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/__init__.py', '/tmp/pip-Py48vV-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/__init__.py', "[Errno 1] Operation not permitted: '/tmp/pip-Py48vV-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/__init__.py'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/__init__.pyc', '/tmp/pip-Py48vV-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/__init__.pyc', "[Errno 1] Operation not permitted: '/tmp/pip-Py48vV-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/__init__.pyc'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/markers.py', '/tmp/pip-Py48vV-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/markers.py', "[Errno 1] Operation not permitted: '/tmp/pip-Py48vV-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/markers.py'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/markers.pyc', '/tmp/pip-Py48vV-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/markers.pyc', "[Errno 1] Operation not permitted: '/tmp/pip-Py48vV-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/markers.pyc'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib', '/tmp/pip-Py48vV-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib', "[Errno 1] Operation not permitted: '/tmp/pip-Py48vV-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib'")]
You are using pip version 7.1.2, however version 9.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
```

bridadan, 2017-04-26:
Hi @dhamukrish, thanks for posting your logs! It looks like this may be a permissions issue.

Could you post the log outputs of the following please?

1. `python --version`
2. `pip --version`
3. `pip list`

Once you have the logs, I'd recommend trying the `pip install` call you did above again, though it seems strange that you need to use `sudo` to do this.

```
cd mbed-os-example-blinky
pip install -r mbed-os/requirements.txt
```

dhamukrish, 2017-04-27:
@bridadan Thanks for your advise.
Please find log below However, I have tried running even in sudo already and provided fresh log below as well.

**```python --version```**
```Python 2.7.11```

**```pip --version```**
```pip 7.1.2 from /Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages (python 2.7)```

**```pip list```**
```altgraph (0.10.2)
appdirs (1.4.3)
backports.ssl-match-hostname (3.5.0.1)
bdist-mpkg (0.5.0)
bonjour-py (0.3)
colorama (0.3.8)
enum34 (1.1.6)
fasteners (0.14.1)
future (0.16.0)
intelhex (2.1)
Jinja2 (2.9.6)
junit-xml (1.7)
macholib (1.5.1)
MarkupSafe (1.0)
matplotlib (1.3.1)
mbed-cli (1.1.1)
mbed-ls (1.2.12)
mercurial (4.2rc0)
modulegraph (0.10.4)
monotonic (1.3)
numpy (1.8.0rc1)
packaging (16.8)
pip (7.1.2)
prettytable (0.7.2)
py2app (0.7.3)
pyobjc-core (2.5.1)
pyobjc-framework-Accounts (2.5.1)
pyobjc-framework-AddressBook (2.5.1)
pyobjc-framework-AppleScriptKit (2.5.1)
pyobjc-framework-AppleScriptObjC (2.5.1)
pyobjc-framework-Automator (2.5.1)
pyobjc-framework-CFNetwork (2.5.1)
pyobjc-framework-Cocoa (2.5.1)
pyobjc-framework-Collaboration (2.5.1)
pyobjc-framework-CoreData (2.5.1)
pyobjc-framework-CoreLocation (2.5.1)
pyobjc-framework-CoreText (2.5.1)
pyobjc-framework-DictionaryServices (2.5.1)
pyobjc-framework-EventKit (2.5.1)
pyobjc-framework-ExceptionHandling (2.5.1)
pyobjc-framework-FSEvents (2.5.1)
pyobjc-framework-InputMethodKit (2.5.1)
pyobjc-framework-InstallerPlugins (2.5.1)
pyobjc-framework-InstantMessage (2.5.1)
pyobjc-framework-LatentSemanticMapping (2.5.1)
pyobjc-framework-LaunchServices (2.5.1)
pyobjc-framework-Message (2.5.1)
pyobjc-framework-OpenDirectory (2.5.1)
pyobjc-framework-PreferencePanes (2.5.1)
pyobjc-framework-PubSub (2.5.1)
pyobjc-framework-QTKit (2.5.1)
pyobjc-framework-Quartz (2.5.1)
pyobjc-framework-ScreenSaver (2.5.1)
pyobjc-framework-ScriptingBridge (2.5.1)
pyobjc-framework-SearchKit (2.5.1)
pyobjc-framework-ServiceManagement (2.5.1)
pyobjc-framework-Social (2.5.1)
pyobjc-framework-SyncServices (2.5.1)
pyobjc-framework-SystemConfiguration (2.5.1)
pyobjc-framework-WebKit (2.5.1)
[... 69 lines cut ...]
Requirement already satisfied (use --upgrade to upgrade): pyparsing in /System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python (from packaging>=16.8->setuptools>=19.0->hidapi->pyOCD>=0.8.1a1->mbed-host-tests>=1.1.2->-r mbed-os/requirements.txt (line 10))
Installing collected packages: setuptools, hidapi, pyOCD, mbed-host-tests, lockfile, funcsigs, pbr, mock, mbed-greentea, beautifulsoup4, fuzzywuzzy
  Found existing installation: setuptools 18.5
    Uninstalling setuptools-18.5:
Exception:
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pip/basecommand.py", line 211, in main
    status = self.run(options, args)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pip/commands/install.py", line 311, in run
    root=options.root_path,
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pip/req/req_set.py", line 640, in install
    requirement.uninstall(auto_confirm=True)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pip/req/req_install.py", line 716, in uninstall
    paths_to_remove.remove(auto_confirm)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pip/req/req_uninstall.py", line 125, in remove
    renames(path, new_path)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pip/utils/__init__.py", line 315, in renames
    shutil.move(old, new)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/shutil.py", line 299, in move
    copytree(src, real_dst, symlinks=True)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/shutil.py", line 208, in copytree
    raise Error, errors
Error: [('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/__init__.py', '/tmp/pip-tVNdaY-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/__init__.py', "[Errno 1] Operation not permitted: '/tmp/pip-tVNdaY-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/__init__.py'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/__init__.pyc', '/tmp/pip-tVNdaY-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/__init__.pyc', "[Errno 1] Operation not permitted: '/tmp/pip-tVNdaY-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/__init__.pyc'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/markers.py', '/tmp/pip-tVNdaY-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/markers.py', "[Errno 1] Operation not permitted: '/tmp/pip-tVNdaY-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/markers.py'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/markers.pyc', '/tmp/pip-tVNdaY-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/markers.pyc', "[Errno 1] Operation not permitted: '/tmp/pip-tVNdaY-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/markers.pyc'"),  [... cut ...] /System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib', '/tmp/pip-tVNdaY-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib', "[Errno 1] Operation not permitted: '/tmp/pip-tVNdaY-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib'")]
You are using pip version 7.1.2, however version 9.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.```

bridadan, 2017-04-27:
Hi @dhamukrish, From the log output it appears to be a pip issue, not an mbed-cli issue. Specifically this line in your last log suggests upgrading pip:

```
You are using pip version 7.1.2, however version 9.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
```

I'd recommend running `pip install --upgrade pip` as suggested and see if that helps with the error.

dhamukrish, 2017-04-27:
Hi @bridadan,
I did already and error remain same.

bridadan, 2017-04-27:
@dhamukrish Ok, well it seems like you have some kind of package issue, specifically with the `setuptools` package. It looks like it's trying to uninstall it which seems strange:

```
Found existing installation: setuptools 18.5
    Uninstalling setuptools-18.5:
```

I found this stack overflow question about it: http://stackoverflow.com/questions/33234665/upgrading-setuptools-on-osx-el-capitan

I'd recommend searching/googling how to solve this particular issue as this is a generic python package issue instead of an issue with mbed CLI.

dhamukrish, 2017-04-28:
@bridadan This helped. Thanks for your direction.
However, I haven't installed stock Python rather I have installed all the missing package individually with sudo.

More over this is not mbed-cli issue.

Thanks again.

closed 2017-04-28 by dhamukrish