ARMmbed/mbed-cli #486: Can't compile on local machine but fine on online compiler

peterpan666, 2017-04-26:
Hi, I've just installed the mbed-cli on my Macbook but can't the project to compile.

It's compiling great with the online compiler though but on my local machine I've got

```
[DEBUG] Output: ./mbed/platform.h:21:20: fatal error: device.h: No such file or directory
```

I've tried to update the librairies and all but nothing seems to work 😔

I'm with `gcc-arm-none-eabi-4_9-2015q2` toolchain and the NRF51_DK target.

theotherjimmy, 2017-04-26:
Hi @peterpan666,

I would like to help you out with this, but I don't have enough information to do that. 
Could you point me at the program you are trying to compile?
Could you tell me which target you are trying to compile for?
Could you list the versions of mbed-cli and mbed-os?

peterpan666, 2017-04-27:
Hi and thanks for your answer! Here we go :

- I can't show you the entire program, it is private and not mine 😕
- The target is the NRF51_DK
- The mbed-cli is `version 1.1.1` I'm not really sure using the mbed-os, how can I see which version it is?

theotherjimmy, 2017-04-27:
> The `mbed-cli` is version 1.1.1 I'm not really sure using the mbed-os, how can I see which version it is?

```bash
mbed ls
```
should do the trick

peterpan666, 2017-04-27:
This is the output of `mbed ls`

```
|- BLE_MIDI (2b71bfbaa458)
|  |- BLE_API (8c645f5694b2)
|  |- mbed (487b796308b0)
|  `- nRF51822 (bdc690669431)
`- LSM9DS1_Library (87d535bf8c53)
```

theotherjimmy, 2017-04-27:
Thanks for the info. You are using the mbed hal revision 96, which is quite old.

peterpan666, 2017-04-27:
Yeah I believe it's due to the `BLE_MIDI` dependency but still, I don't get why I could compile the project online and not on my local machine 😕

theotherjimmy, 2017-04-27:
Agreed. It should compile in both. I'm going to try to reproduce the error on my end. If I can, then I can give you some insight on why it's failing.

theotherjimmy, 2017-04-27:
I created a test project with an empty main to test the BLE_MIDI + sub-library structure.
`mbed ls`
```
testing-BLE-MIDI (None)
`- BLE_MIDI (2b71bfbaa458)
   |- BLE_API (8c645f5694b2)
   |- mbed (487b796308b0)
   `- nRF51822 (bdc690669431)
```
`mbed compile -t gcc_arm -m nrf51_dk` 
```
Building project testing-BLE-MIDI (NRF51_DK, GCC_ARM)
Scan: .
Scan: mbed
Scan: env
Compile [  3.2%]: GapAdvertisingParams.cpp
Compile [  6.5%]: UUID.cpp
Compile [  9.7%]: BLEDevice.cpp
Compile [ 12.9%]: GattService.cpp
Compile [ 16.1%]: GapAdvertisingData.cpp
Compile [ 19.4%]: DFUService.cpp
Compile [ 22.6%]: UARTService.cpp
Compile [ 25.8%]: btle_advertising.cpp
[Warning] assertion.h@83,1: multi-line comment [-Wcomment]
[Warning] assertion.h@86,1: multi-line comment [-Wcomment]
[Warning] ble_gap.h@722,42: 'uint32_t sd_ble_gap_address_set(uint8_t, const ble_gap_addr_t*)' defined but not used [-Wunused-function]
Compile [ 29.0%]: btle_gap.cpp
[Warning] assertion.h@83,1: multi-line comment [-Wcomment]
[Warning] assertion.h@86,1: multi-line comment [-Wcomment]
[Warning] ble_gap.h@722,42: 'uint32_t sd_ble_gap_address_set(uint8_t, const ble_gap_addr_t*)' defined but not used [-Wunused-function]
Compile [ 32.3%]: app_gpiote.c
Compile [ 35.5%]: custom_helper.cpp
[Warning] assertion.h@83,1: multi-line comment [-Wcomment]
[Warning] assertion.h@86,1: multi-line comment [-Wcomment]
[Warning] ble_gap.h@722,42: 'uint32_t sd_ble_gap_address_set(uint8_t, const ble_gap_addr_t*)' defined but not used [-Wunused-function]
Compile [ 38.7%]: crc16.cpp
Compile [ 41.9%]: app_scheduler.c
Compile [ 45.2%]: hci_mem_pool.c
Compile [ 48.4%]: nRF51Gap.cpp
[Warning] assertion.h@83,1: multi-line comment [-Wcomment]
[Warning] assertion.h@86,1: multi-line comment [-Wcomment]
[Warning] nRF51Gap.cpp@226,12: enumeration value 'LOCAL_HOST_TERMINATED_CONNECTION' not handled in switch [-Wswitch]
Compile [ 51.6%]: btle.cpp
[Warning] assertion.h@83,1: multi-line comment [-Wcomment]
[Warning] assertion.h@86,1: multi-line comment [-Wcomment]
[Warning] nordic_common.h@36,9: comparison of unsigned expression < 0 is always false [-Wtype-limits]
[Warning] nordic_common.h@36,9: comparison of unsigned expression < 0 is always false [-Wtype-limits]
Compile [ 54.8%]: ble_bondmngr.cpp
Compile [ 58.1%]: ble_advdata_parser.cpp
[Warning] ble_gap.h@722,42: 'uint32_t sd_ble_gap_address_set(uint8_t, const ble_gap_addr_t*)' defined but not used [-Wunused-function]
Compile [ 61.3%]: BLEMIDI.cpp
[Warning] BLEMIDI.cpp@268,24: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
[Warning] BLEMIDI.cpp@432,0: suggest parentheses around arithmetic in operand of '|' [-Wparentheses]
[Warning] BLEMIDI.cpp@441,0: suggest parentheses around arithmetic in operand of '|' [-Wparentheses]
Compile [ 64.5%]: nRF51822n.cpp
[Warning] nRF51822n.h@20,0: "NRF51" redefined
[Warning] assertion.h@83,1: multi-line comment [-Wcomment]
[Warning] assertion.h@86,1: multi-line comment [-Wcomment]
[Warning] ble_gap.h@722,42: 'uint32_t sd_ble_gap_address_set(uint8_t, const ble_gap_addr_t*)' defined but not used [-Wunused-function]
Compile [ 67.7%]: ble_debug_assert_handler.cpp
Compile [ 71.0%]: bootloader_util_arm.c
Compile [ 74.2%]: ble_advdata.cpp
[Warning] ble_gap.h@722,42: 'uint32_t sd_ble_gap_address_set(uint8_t, const ble_gap_addr_t*)' defined but not used [-Wunused-function]
Compile [ 77.4%]: ble_srv_common.cpp
[Warning] ble_gap.h@722,42: 'uint32_t sd_ble_gap_address_set(uint8_t, const ble_gap_addr_t*)' defined but not used [-Wunused-function]
Compile [ 80.6%]: ble_error_log.cpp
Compile [ 83.9%]: nRF51GattServer.cpp
[Warning] assertion.h@83,1: multi-line comment [-Wcomment]
[Warning] assertion.h@86,1: multi-line comment [-Wcomment]
[Warning] ble_gap.h@722,42: 'uint32_t sd_ble_gap_address_set(uint8_t, const ble_gap_addr_t*)' defined but not used [-Wunused-function]
Compile [ 87.1%]: main.cpp
Compile [ 90.3%]: dfu_app_handler.c
Compile [ 93.5%]: softdevice_handler.cpp
[Warning] nrf_soc.h@348,32: 'uint32_t sd_mutex_new(nrf_mutex_t*)' defined but not used [-Wunused-function]
Compile [ 96.8%]: pstorage.cpp
[Warning] nrf_soc.h@348,32: 'uint32_t sd_mutex_new(nrf_mutex_t*)' defined but not used [-Wunused-function]
Compile [100.0%]: ble_conn_params.cpp
[Warning] ble_gap.h@722,42: 'uint32_t sd_ble_gap_address_set(uint8_t, const ble_gap_addr_t*)' defined but not used [-Wunused-function]
Link: testing-BLE-MIDI
Elf2Bin: testing-BLE-MIDI
+-----------+-------+-------+------+
| Module    | .text | .data | .bss |
+-----------+-------+-------+------+
| Fill      |    37 |     0 |   22 |
| Misc      | 15347 |   128 | 1438 |
| Subtotals | 15384 |   128 | 1460 |
+-----------+-------+-------+------+
Allocated Heap: 2048 bytes
Allocated Stack: 2048 bytes
Total Static RAM memory (data + bss): 1588 bytes
Total RAM memory (data + bss + heap + stack): 5684 bytes
Total Flash memory (text + data + misc): 15512 bytes

Image: ./BUILD/nrf51_dk/gcc_arm/testing-BLE-MIDI.hex
```
Looks fine. 

I don't know what's going wrong here.

peterpan666, 2017-04-28:
Sigh and with the same libraries version 😔
Ok I'll try to find something missing and put it back here if I find anything.
Thanks for your help and time!

peterpan666, 2017-04-28:
Wich version of GCC ARM are you using?

theotherjimmy, 2017-04-28:
You're welcome.

I'm just glad that you got it working :D