ARMmbed/mbed-cli #487: Duplicate optimization settings on IAR EWARM export

ASealion, 2017-05-02:
tested on 7.7 and 8.11

**Steps to reproduce**
After exporting a project from the online compiler is is not possible to compile the project. 
Since the project options on the "Optimizer" are set to "High" and there is a extra Option "-Oh".

This leads to the compiler message:
[Su004]: Option -O can only occur once

0xc0170, 2017-05-02:
cc @theotherjimmy

theotherjimmy, 2017-05-02:
Whoops. Thanks for finding this. The python that is run when you export is here: https://github.com/ARMmbed/mbed-os/tree/master/tools/export/iar This may not be the only duplicate option. You are welcome to submit a PR to mbed-os @ASealion. I can help out, or make the PR.

theotherjimmy, 2017-05-03:
Good call on the labels @screamerbg.

I'm going to take a look today, @ASealion. I'll mention the PR I create here.

theotherjimmy, 2017-05-05:
Looks like [this line](https://github.com/ARMmbed/mbed-os/blob/master/tools/export/iar/__init__.py#L117) needs to check for `-Oh`

theotherjimmy, 2017-05-09:
See https://github.com/ARMmbed/mbed-os/pull/4293 for the fix

Val-Apana, 2017-05-10:
In IAR: Project -> Options-> C/C++ Compiler: “Extra Options” tab, scroll down and delete the “-Oh” line.

theotherjimmy, 2017-05-10:
Yup. That's what the PR I mentioned prevents you from having to do.

closed 2017-05-15 by 0xc0170