ARMmbed/mbed-cli #490: "Makefile" added into GIT repo excludes

kuggenhoffen, 2017-05-23:
It seems that mbed-cli adds `Makefile` (and others) to `.git/info/exclude`. What's the reason to force repository wide ignore of those? 

I presume it is because of project export, if so I think these ignores could be more specific so that user is allowed to have such files in their repository but still ignoring generated project files.

bridadan, 2017-05-24:
@kuggenhoffen Yes I believe this is due to project export. Would you suggest something like we only ignore the Makefile in the root of the project? What are your thoughts on this?

kuggenhoffen, 2017-05-29:
I'm not too familiar with the export functionality, but perhaps something like have the project file(s) exported to known location / name and only ignore the generated files in that location would work. That would keep repository clean of autogenerated export files while allowing user to keep track of their own makefiles and/or project files.

bridadan, 2017-05-30:
@kuggenhoffen The problem with `Makefile`s is that (to my knowledge) they tend to only be named one thing 😄

However, exported project files are always put in the project root, never anywhere else. It may be possible to ignore ONLY this location, however there may be other things that I'm not aware of that could complicate this.

Any thoughts @theotherjimmy?

theotherjimmy, 2017-05-30:
We could ignore only the root Makefile. I can't see how that would negatively impact things. We should try it out and see if it's a Bad Thing™ !

screamerbg, 2017-06-21:
cc @0xc0170

0xc0170, 2017-06-21:
> We could ignore only the root Makefile. I can't see how that would negatively impact things. We should try it out and see if it's a Bad Thing™ !

Can you test that patch and send it upstream @theotherjimmy ?

theotherjimmy, 2017-06-21:
Yup can do.

PR #515 (2017-06-23): Ignore root makefiles only
Resolves #490
closed 2017-06-23 by screamerbg