ARMmbed/mbed-cli #491: How to import from a specific branch of a gitlab repository

binary-nerd, 2017-05-24:
I have a question related to the workflow using mbed-cli

Here is what I did.
1. I went into my embedded project which was for arduino.
2. I creatd a new branch called mbed.
3. I cleaned up everything that is existing
4. `mbed new --scm none --program .`  -> This downloads everything including mbed-os
5. I created `main.cpp` and wrote a program
6. I pushed the new branch to my repository. Interestingly the mbed-os did not get uploaded.

So, I want to use this project on another machine.
What are the steps to use it back?

I thought i can `mbed import` it using the github URL and it installs mbed-os along with my code. But, I dont see any option to specify a branch name while importing from a gitlab link.

### Questions:
1. What is the recommended workflow with mbed-cli?
2. How to download a project from a specific branch of a gitlab repository

bridadan, 2017-05-24:
I believe you should be able to do an `mbed import` of your git repo, then do a `mbed update [branch_name]` in the project directory. Here's the relevant portion of the docs: https://github.com/ARMmbed/mbed-cli#updating-a-program

binary-nerd, 2017-05-25:
Thanks @bridadan 
I made a small make file which would download the mbed-os sources and few dependencies. For now, I am good with this approach.

closed 2017-05-25 by binary-nerd