ARMmbed/mbed-cli #492: cstddef: No such file or directory Error with mbed-os-example-blinky project

binary-nerd, 2017-05-25:
Compiling mbed-os-example-blinky gives me the following errors.

```
mbed-os-example-blinky $ mbed compile
Building project mbed-os-example-blinky (NUCLEO_F091RC, GCC_ARM)
Scan: .
Scan: FEATURE_BLE
Scan: FEATURE_COMMON_PAL
Scan: FEATURE_UVISOR
Scan: FEATURE_LWIP
Scan: FEATURE_STORAGE
Scan: FEATURE_NANOSTACK
Scan: FEATURE_LOWPAN_BORDER_ROUTER
Scan: FEATURE_ETHERNET_HOST
Scan: FEATURE_THREAD_END_DEVICE
Scan: FEATURE_NANOSTACK_FULL
Scan: FEATURE_LOWPAN_HOST
Scan: FEATURE_THREAD_ROUTER
Scan: FEATURE_THREAD_BORDER_ROUTER
Scan: FEATURE_LOWPAN_ROUTER
Scan: mbed
Scan: env
Compile [  0.4%]: AnalogIn.cpp
[Fatal Error] platform.h@22,19: cstddef: No such file or directory
[ERROR] In file included from ./mbed-os/drivers/AnalogIn.h:19:0,
                 from ./mbed-os/drivers/AnalogIn.cpp:17:
./mbed-os/platform/platform.h:22:19: fatal error: cstddef: No such file or directory
 #include <cstddef>
                   ^
compilation terminated.

[mbed] ERROR: "/usr/bin/python" returned error code 1.
[mbed] ERROR: Command "/usr/bin/python -u /home/sandeep/temp_backup/mbed-os-example-blinky/mbed-os/tools/make.py -t GCC_ARM -m NUCLEO_F091RC --source . --build ./BUILD/NUCLEO_F091RC/GCC_ARM" in "/home/sandeep/temp_backup/mbed-os-example-blinky"
```

I know there is some setup issue. But I am not able to figure it out what it is!

0xc0170, 2017-05-25:
@binary-nerd Whats your GCC setup in your system (in your path) ? Is it gcc arm, not other distribution? what version ? 

Run ``arm-none-eabi-gcc --version``

binary-nerd, 2017-05-26:
Yes it is `GCC_ARM`

```
arm-none-eabi-gcc (15:4.9.3+svn231177-1) 4.9.3 20150529 (prerelease)
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Also there does not seem to be any other ARM binaries in my `/usr/bin`

```
mbed-os-example-blinky $ ls /usr/bin/ | grep gcc
arm-none-eabi-gcc
arm-none-eabi-gcc-4.9.3
arm-none-eabi-gcc-ar
arm-none-eabi-gcc-nm
arm-none-eabi-gcc-ranlib
c89-gcc
c99-gcc
gcc
gcc-5
gcc-ar
gcc-ar-5
gcc-nm
gcc-nm-5
gcc-ranlib
gcc-ranlib-5
x86_64-linux-gnu-gcc
x86_64-linux-gnu-gcc-5
x86_64-linux-gnu-gcc-ar
x86_64-linux-gnu-gcc-ar-5
x86_64-linux-gnu-gcc-nm
x86_64-linux-gnu-gcc-nm-5
x86_64-linux-gnu-gcc-ranlib
x86_64-linux-gnu-gcc-ranlib-5
mbed-os-example-blinky $ ls /usr/bin/ | grep g++
arm-none-eabi-g++
g++
g++-5
x86_64-linux-gnu-g++
x86_64-linux-gnu-g++-5
```

0xc0170, 2017-05-26:
Can you update (it writes prerelease) your gcc arm? If you google you will find that often using official gcc arm should fix this problem or having cpp file instead of c - two most common problems to get this type of error (it was already reported even here on github, please look for it).

binary-nerd, 2017-07-03:
Thanks @0xc0170 

- I downloaded the latest available toolchain from [ARM developers site](https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads)
- Set GCC_ARM_PATH accordingly (`mbed config GCC_ARM_PATH "/home/binarynerd/sw/toolchain/gcc-arm-none-eabi-6-2017-q1-update/bin`")

Then it started to work. Thank you.

closed 2017-07-03 by binary-nerd