ARMmbed/mbed-cli #494: Understanding the build system of mbed

binary-nerd, 2017-05-30:
I dont see any makefile sought of thing where i can define what files to include.

What is the build system used by `mbed`?

For example, I am trying to include a BLE library and I am facing following error :
```
Compile [  1.4%]: BlueNRGDiscoveredCharacteristic.cpp
[Fatal Error] BlueNRGDiscoveredCharacteristic.h@20,42: ble/DiscoveredCharacteristic.h: No such file or directory
[ERROR] In file included from ../ble-x-nucleo-idb0xa1/source/BlueNRGDiscoveredCharacteristic.cpp:17:0:
../ble-x-nucleo-idb0xa1/x-nucleo-idb0xa1/BlueNRGDiscoveredCharacteristic.h:20:42: fatal error: ble/DiscoveredCharacteristic.h: No such file or directory
```
I would like to check if the includes path include the BLE path or not. How can i check that?

By default all the files are being included. Do we have a precise controle over what files and libraries are included?

bridadan, 2017-05-30:
@binary-nerd all files in the project directory are included unless you specifically ignore them with a `.mbedignore` file, which is documented here: https://docs.mbed.com/docs/mbed-os-handbook/en/latest/advanced/mb

What is the structure of your app? What library are you using?

binary-nerd, 2017-05-30:
Thanks @bridadan 
I was able to remove the errors by adding BLE_API to the project.

But this step was not intuitive to me. BLE_API was already present in mbed-os, so i initially thought it would include it from htere, but that did not happen.

If you want to include header path or declare some preprocessor variables, do we have a way to do that in mbed?

bridadan, 2017-05-31:
@binary-nerd Is this project public? If it is, can you post a link? That way we can make sure the tools are behaving how they are supposed to.

But yes there is a BLE folder in mbed-os, though this is only for boards that have native BLE support (ex. NRF51_DK).

binary-nerd, 2017-06-05:
Thanks @bridadan 
The project is not public. Actually my question is whether we have any finer controle on the build system.
like passing preprocessor flags etc.. like what we can do in a traditional linux makefile.

I am closing the issue for now as my original problem is resolved.  We can stil continue the discussion.

Thank you.

closed 2017-06-05 by binary-nerd

theotherjimmy, 2017-06-05:
You can pass preproccessor flags to `mbed compile` in the same way you would pass them to the compiler: `-DFOO=bar`

bridadan, 2017-06-05:
@binary-nerd You can do as @theotherjimmy suggested. You are also welcome to export to a Makefile if you are more comfortable with that. Assuming you're using the `GCC_ARM` compiler, the mbed CLI command would be.

```
 mbed export -i make_gcc_arm -m <INSERT YOUR TARGET HERE>
```

At which point you should be able to just run `make`

pan-, 2017-06-05:
@binary-nerd To include BLE api in your project you have to enable the `BLE` feature in your application. 
You can look at `mbed_app.json` files in mbed-os-example-ble as an example (for instance [here](https://github.com/ARMmbed/mbed-os-example-ble/blob/master/BLE_BatteryLevel/mbed_app.json#L4))