ARMmbed/mbed-cli #496: module python missing

rspelta, 2017-06-07:
Hi all,
I'm installing from a virtual machine (windows 7 os, it is clean, just installed) the mbed-cli and I'm having a problem. After the installation following the video https://www.youtube.com/watch?v=cM0dFoTuU14 I tried to compile the project but I get this error:
> mbed compile
Traceback (most recent call last):
  File "C:\Users\mbed\Desktop\mbed-os-sensor-node\mbed-os\tools\make.py", line 31, in <module>
    from tools.utils import args_error
  File "C:\Users\mbed\Desktop\mbed-os-sensor-node\mbed-os\tools\utils.py", line 31, in <module>
    from intelhex import IntelHex
ImportError: No module named intelhex
[mbed] Auto-installing missing Python modules...
[mbed] WARNING: Unable to auto-install required Python modules.
[mbed] WARNING: -----------------------------------------------------------------
[mbed] WARNING: The mbed OS tools in this program require the following Python modules: colorama, pyserial, prettytable, jinja2, intelhex, junit_xml, pyyaml, requests, mbed_ls, mbed_host_tests, mbed_greentea, beautifulsoup4, fuzzywuzzy
[mbed] WARNING: You can install all missing modules by running "pip install -r requirements.txt" in "C:\Users\mbed\Desktop\mbed-os-sensor-node\mbed-os"
[mbed] ERROR: "c:\python27\python.exe" returned error code 1.
[mbed] ERROR: Command "c:\python27\python.exe -u C:\Users\mbed\Desktop\mbed-os-sensor-node\mbed-os\tools\make.py -t GCC_ARM -m NUCLEO_L476RG --source . --build .\BUILD\NUCLEO_L476RG\GCC_ARM" in "C:\Users\mbed\Desktop\mbed-os-sensor-node"
---

I resolved the problem installing manually the modules with this command:

pip install IntelHex jinja2 bs4 fuzzywuzzy prettytable colorama pyserial junit_xml pyyaml requests mbed_ls

example:
> pip install IntelHex
Collecting IntelHex
  Using cached intelhex-2.1-py2.py3-none-any.whl
Installing collected packages: IntelHex
Successfully installed IntelHex-2.1

but the modules **mbed_host_tests** and **mbed_greentea** are missing. 
example:
> pip install **mbed_host_tests**
Collecting mbed_host_tests
  Using cached mbed-host-tests-1.1.8.tar.gz
Collecting PySerial>=3.0 (from mbed_host_tests)
  Using cached pyserial-3.3-py2.py3-none-any.whl
Collecting PrettyTable>=0.7.2 (from mbed_host_tests)
  Using cached prettytable-0.7.2.zip
Collecting requests (from mbed_host_tests)
  Using cached requests-2.17.3-py2.py3-none-any.whl
Collecting mbed-ls>=1.0.0 (from mbed_host_tests)
  Using cached mbed-ls-1.2.13.tar.gz
Collecting pyOCD>=0.8.1a1 (from mbed_host_tests)
  Using cached pyOCD-0.8.1a1.tar.gz
    Complete output from command python setup.py egg_info:
    Download error on https://pypi.python.org/simple/setuptools_scm/: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:661) -- Some packages may not be found!
    Download error on https://pypi.python.org/simple/setuptools-scm/: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:661) -- Some packages may not be found!
    Couldn't find index page for 'setuptools_scm' (maybe misspelled?)
    Download error on https://pypi.python.org/simple/: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:661) -- Some packages may not be found!
    No local packages or working download links found for setuptools_scm!=1.5.3,!=1.5.4
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "c:\users\mbed\appdata\local\temp\pip-build-iugdm1\pyOCD\setup.py", line 66, in <module>
        include_package_data=True,  # include files from MANIFEST.in
      File "c:\python27\lib\distutils\core.py", line 111, in setup
        _setup_distribution = dist = klass(attrs)
      File "c:\python27\lib\site-packages\setuptools\dist.py", line 315, in __init__
        self.fetch_build_eggs(attrs['setup_requires'])
      File "c:\python27\lib\site-packages\setuptools\dist.py", line 361, in fetch_build_eggs
        replace_conflicting=True,
      File "c:\python27\lib\site-packages\pkg_resources\__init__.py", line 850, in resolve
        dist = best[req.key] = env.best_match(req, ws, installer)
      File "c:\python27\lib\site-packages\pkg_resources\__init__.py", line 1122, in best_match
        return self.obtain(req, installer)
      File "c:\python27\lib\site-packages\pkg_resources\__init__.py", line 1134, in obtain
        return installer(requirement)
      File "c:\python27\lib\site-packages\setuptools\dist.py", line 429, in fetch_build_egg
        return cmd.easy_install(req)
      File "c:\python27\lib\site-packages\setuptools\command\easy_install.py", line 659, in easy_install
        raise DistutilsError(msg)
    distutils.errors.DistutilsError: Could not find suitable distribution for Requirement.parse('setuptools_scm!=1.5.3,!=1.5.4')
Command "python setup.py egg_info" failed with error code 1 in c:\users\mbed\appdata\local\temp\pip-build-iugdm1\pyOCD\

The only way to install them is to download from a repository using the browser then install them from the zip file.

Two days ago I didn't have this problem but since yesterday afternoon the problem is started. I tried 3 times with 3 virtual machine clean with installed Windows 7. And for three times I having this problem.

Any suggestion?
Thank you

Roberto Spelta

bridadan, 2017-06-07:
Thanks for reporting this. However to my knowledge we haven't made any changes to the packages you mentioned.

From the error log it looks like you may be having an SSL issue. This isn't something we can fix.

I did a bit of searching and found this discussion thread, maybe something there will help you? https://github.com/pypa/pip/issues/3415

rspelta, 2017-06-08:
Thank you for the reply, the only way I findout to resolve this problem is to install in this order the packages:
pip install mbed-cli IntelHex jinja2 bs4 fuzzywuzzy prettytable colorama pyserial junit_xml pyyaml requests mbed_ls
pip install setuptools_scm
pip install mbed-host-tests
pip intall mbed-greentea

Seems that setuptools_scm is a package to install in order to avoid that error.

bridadan, 2017-06-12:
@rspelta Thanks for posting your findings. We don't have control over those packages unfortunately. I'm gonna close this issue for now, if others run into this we can look into this issue in more depth.

closed 2017-06-12 by bridadan