ARMmbed/mbed-cli #500: BLEProtocol.h@22,21: algorithm: No such file or directory

pmcalabrese, 2017-06-08:
Hello,
I'm trying to compile https://developer.mbed.org/teams/microbit/code/microbit-hello-world/ with mbed-cli.
I'm using the latest mbed-cli.

when i run the command

```
mbed compile -t GCC_ARM -m nrf51_microbit
```

i get the following output

```
Building project microbit-hello-world (NRF51_MICROBIT, GCC_ARM)
Scan: .
Scan: mbed
Scan: env
Compile [ 21.2%]: ble_dfu.c
[Fatal Error] BLEProtocol.h@22,21: algorithm: No such file or directory
[ERROR] In file included from ./microbit/microbit-dal/BLE_API/ble/Gap.h:20:0,
                 from ./microbit/microbit-dal/BLE_API/ble/ble.h:21,
                 from ./microbit/microbit-dal/nRF51822/nrf51-sdk/source/nordic_sdk/components/ble/ble_services/ble_dfu/ble_dfu.h:53,
                 from ./microbit/microbit-dal/nRF51822/nrf51-sdk/source/nordic_sdk/components/ble/ble_services/ble_dfu/ble_dfu.c:33:
./microbit/microbit-dal/BLE_API/ble/BLEProtocol.h:22:21: fatal error: algorithm: No such file or directory
 #include <algorithm>
                     ^
compilation terminated.

[mbed] ERROR: "/usr/local/opt/python/bin/python2.7" returned error code 1.
[mbed] ERROR: Command "/usr/local/opt/python/bin/python2.7 -u /Users/mca/dev/microbit/microbit-hello-world/.temp/tools/make.py -t GCC_ARM -m nrf51_microbit --source . --build ./BUILD/nrf51_microbit/GCC_ARM" in "/Users/mca/dev/microbit/microbit-hello-world"
```

The same program compiles on the online editor without any errors.

Any suggestion or help is welcome.

screamerbg, 2017-06-21:
CC @pan-

pan-, 2017-06-21:
@pmcalabrese This error is in the Lancaster University fork of BLE_API. It has been fixed a long time ago in mbed OS. 

What is happening is simple, the mbed build system add to the include path any folder reachable by the target (in this case the NRF51_MICROBIT). In this case, the compiler try to includes the file `ble.h` but unfortunately there is two files named `ble.h` in the include path: one defined by `BLE_API` the other defined by the `nrf51-sdk`. The file `ble_dfu.c` needs `ble.h` from the nrf51 sdk but it is the one from `BLE_API` which is included. Therefore it raises a compilation error (`ble.h` from BLE_API is a C++ and not C header). 

The solution we adopted in mbed was to prefix filename in the nrf51 SDK. You can raise this issue or submit a fix in the [microbit fork](https://github.com/lancaster-university/nrf51-sdk).

theotherjimmy, 2017-11-28:
Closing this as a problem with a particular library.

closed 2017-11-28 by theotherjimmy